feram18/led-stock-ticker #1: Flickering

callyb123, 2021-12-05:
Hey Luis,

Thank you so much for this, it's awesome. I noticed some flickering when stock names are too large to display at the top. Does this utilize the adafruit bonnet hat-pwm flags? I ran with these flags and it seemed to make no difference to the flicker.

feram18, 2021-12-06:
Hi! The flickering seems to be due to the rendering of the chart being an expensive operation, as it's rendered pixel by pixel.

I have not had much time to work on this project lately, but I'll be working on a way to improve the performance when the stock name has to scroll. I have also considered getting rid of the full name altogether and switch up the layout a little bit. Updates to come!

callyb123, 2021-12-06:
Yup - I think getting rid of the full name scrolling and reverting to the ticker symbol is most likely the best option

0x253, 2022-01-03:
Hi, for me it works without flickering with the `--led-slowdown-gpio` parameter:

`sudo python3 main.py --led-slowdown-gpio=4 --led-cols=64 --led-rows=32 --led-gpio-mapping=adafruit-hat`

feram18, 2022-10-31:
Performance enhancements in [latest releases](https://github.com/feram18/led-stock-ticker/releases/tag/v0.3.1) (in addition to `--led-slowdown-gpio` flag) should take care of the flickering.

closed 2022-10-31 by feram18