feram18/led-stock-ticker #2: Stocks do not automatically update

callyb123, 2021-12-06:
Stocks do not seem to update and keep the price that they were at when the script was first run.

[image]

feram18, 2021-12-08:
Hi,

The update rate is set to 15 min, the script checks if data should be updated (i.e. 15 mins have passed since data was last updated) after it has rotated through all of the tickers. I set a conservative update rate as I was seeing some issues receiving data from the yfinance library when making a lot of requests too frequently after a few hours, not sure if that was just a personal issue. However, I am considering adding update rate as a configurable option.

Try running the script on debug mode by adding the `--debug` flag to your command. You should be able to see debug messages that look like this when updating tickers' prices.

```
12/07/2021 10:40:30 PM INFO: Checking for update
12/07/2021 10:40:30 PM DEBUG: Fetching new data for MSFT.
12/07/2021 10:40:30 PM DEBUG: Fetching new data for AMC.
```

callyb123, 2021-12-08:
Appreciate your help Luis. I’ll try this and get back to you.

closed 2022-10-31 by feram18