feram18/led-stock-ticker #3: Numpy Value Error

Jordinho, 2021-12-29:
Hello,

I am getting the following error when running the main script:

"ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 44 from C header, got 40 from PyObject"

Can you tell my what version of Numpy you are using?  I was able to get it to work by changing the numpy version but I forgot what I did and formatted my drive and cant figure it out again.

feram18, 2021-12-29:
As listed on the `requirements.txt` file numpy should be on version 1.16.1 or higher.

I currently have 1.16.6 on a Raspberry Pi 4B, and 1.21.0 on a 3B+ and have not encountered this issue.

Which numpy version are you running on?

Jordinho, 2021-12-30:
Thanks for the reply.  I uninstalled the numpy and reinstalled (1.21.5) and that seemed to fix the issue. I tried this before, but I guess the second time it worked.  For reference, this is what worked for me.

Package               Version
--------------------- -----------
altgraph              0.17.2
appdirs               1.4.4
APScheduler           3.8.1
arandr                0.1.10
astroid               2.5.1
asttokens             2.0.4
attrs                 20.3.0
automationhat         0.2.0
beautifulsoup4        4.9.3
blinker               1.4
blinkt                0.1.2
Bottleneck            1.2.1
buttonshim            0.0.2
CacheControl          0.12.10
cairocffi             1.2.0
CairoSVG              2.5.0
Cap1xxx               0.1.3
certifi               2020.6.20
cffi                  1.14.5
chardet               3.0.4
click                 7.1.2
colorama              0.4.4
colorzero             1.1
cryptography          3.3.2
cssselect2            0.3.0
cupshelpers           1.0
cycler                0.10.0
Cython                0.29.26
dbus-python           1.2.16
decorator             5.1.0
defusedxml            0.6.0
distro                1.5.0
docutils              0.16
drumhat               0.1.0
env-canada            0.0.35
envirophat            1.0.0
ephem                 3.7.7.0
et-xmlfile            1.0.1
ExplorerHAT           0.4.2
fastjsonschema        2.15.2
feedparser            6.0.8
Flask                 1.1.2
fourletterphat        0.1.0
future                0.18.2
geocoder              1.38.1
geographiclib         1.52
geojson               2.5.0
geopy                 2.2.0
gpiozero              1.5.1
html5lib              1.1
idna                  2.8
imageio               2.13.5
importlib-metadata    1.6.0
iniconfig             0.0.0
iso6709               0.1.5
isort                 5.6.4
itsdangerous          1.1.0
jdcal                 1.0
jedi                  0.18.0
Jinja2                2.11.3
joblib                0.17.0
kiwisolver            1.3.1
lastversion           2.0.1
lazy-object-proxy     0.0.0
lockfile              0.12.2
logilab-common        1.8.1
lxml                  4.6.3
MarkupSafe            1.1.1
matplotlib            3.3.4
mccabe                0.6.1
microdotphat          0.2.1
more-itertools        4.2.0
mote                  0.0.4
motephat              0.0.3
msgpack               1.0.3
multitasking          0.0.10
mypy                  0.812
mypy-extensions       0.4.3
nameparser            1.0.6
noaa-sdk              0.1.19
nose                  1.3.7
numexpr               2.7.2
numpy                 1.21.5
oauthlib              3.1.0
odfpy                 1.4.1
olefile               0.46
openpyxl              3.0.3
packaging             21.3
pandas                1.3.5
pantilthat            0.0.7
parso                 0.8.1
pexpect               4.8.0
pgzero                1.2
phatbeat              0.1.1
pianohat              0.1.0
picamera              1.13
piglow                1.2.5
pigpio                1.78
Pillow                8.3.2
pip                   20.3.4
pluggy                0.13.0
ply                   3.11
printtools            1.2
prompt-toolkit        3.0.24
psutil                5.8.0
py                    1.10.0
pycairo               1.16.2
pycocotools           2.0.3
pycparser             2.20
pycups                2.0.1
pyfiglet              0.8.post1
pygame                1.9.6
Pygments              2.7.1
PyGObject             3.38.0
pyinotify             0.9.6
PyInstaller           3.6
PyJWT                 1.7.1
pylint                2.7.2
pyOpenSSL             20.0.1
pyowm                 3.2.0
pyparsing             3.0.6
pyserial              3.5b0
pysmbc                1.0.23
PySocks               1.7.1
pytest                6.0.2
python-apt            2.2.1
python-dateutil       2.8.2
python-tsl2591        0.2.0
pytz                  2021.3
pytz-deprecation-shim 0.1.0.post0
PyYAML                6.0
questionary           1.10.0
rainbowhat            0.1.0
ratelim               0.1.6
ratelimit             2.2.1
regex                 2021.11.10
reportlab             3.5.59
requests              2.22.0
requests-futures      1.0.0
requests-oauthlib     1.0.0
responses             0.12.1
rgbmatrix             0.0.1
roman                 2.0.0
RPi.GPIO              0.7.0
RTIMULib              7.2.1
scikit-learn          0.23.2
scipy                 1.6.0
scrollphat            0.0.7
scrollphathd          1.2.1
Send2Trash            1.6.0b1
sense-hat             2.2.0
setuptools            52.0.0
sgmllib3k             1.0.0
simplejson            3.17.2
six                   1.16.0
skywriter             0.0.7
smbus2                0.4.1
sn3218                1.2.7
soupsieve             2.2.1
spidev                3.5
ssh-import-id         5.10
supervisor            4.2.2
tables                3.6.1
thonny                3.3.14
tinycss2              1.0.2
toml                  0.10.1
touchphat             0.0.1
tqdm                  4.62.3
twython               3.8.2
typed-ast             1.4.2
typing-extensions     3.7.4.3
tzdata                2021.5
tzlocal               4.1
unicornhathd          0.0.4
urllib3               1.25.11
wcwidth               0.2.5
webencodings          0.5.1
Werkzeug              1.0.1
wheel                 0.37.1
wrapt                 1.12.1
xcffib                0.8.1
xlwt                  1.3.0
yfinance              0.1.67
zipp                  1.0.0

closed 2022-01-02 by feram18