feram18/led-stock-ticker #4: Illegal instruction

ElPolloLoco007, 2021-12-30:
I just get "Illegal instruction" when i run the command: sudo main.py...

I had rpi-rgb-led-matrix already installed, when i installed this project.

The other project still runs fine. Have no idea what the problem might be

ElPolloLoco007, 2021-12-31:
probably something to do with pi zero

closed 2021-12-31 by ElPolloLoco007

feram18, 2022-01-02:
Sorry for the late reply. Since you had the library already installed, did you use the install script from this project? If not, it could be that one of the dependencies are missing (or not in a supported version). Try installing/updating just the dependencies with `sudo pip3 install -r requirements.txt` if that's the case.

I'm aware that the Pi Zero has some issues with the rpi-rgb-led-matrix library, but as far as I know it's only performance issues. Unfortunately I don't own a Pi Zero so I can't confirm if hardware is the issue.