feram18/led-stock-ticker #5: Nice project!

0x253, 2022-01-04:
Hi,

I want only to say that I use this stock ticker for my RGB LED display, and it works great! I use it on a raspberry pi 4 with a Adafruit HAT and a 32x64 LED display. I adapted the code to display the date for a different locale. It would be nice if some of the constants can be defined in the config, so that the source must not be touched:

* Rotation duration (I use 5 secs, 10 secs is too long for me)
* Different date format string (`%a, %d. %b`)
* Different locale for the date (by setting `locale.setlocale(locale.LC_TIME, 'DE_de')` in the Data class)

Everything else works really great!

feram18, 2022-01-04:
Thank you so much for the feedback!

These are great suggestions! I will be adding the rotation rate as a configurable option, and various date format options as well.

As for the locale, it should be picking up your local date/time from your Raspberry Pi's settings. Try `sudo raspi-config` > Localisation Options > Timezone and check if you have the appropriate selection. Not sure if I'm misunderstanding something, but when I change to another timezone it seems get the appropriate data & time for that timezone.

gcacciola, 2022-01-04:
Just commenting to give you a thumbs up and keep you motivated through the project.

I tested a lot of different Led tickers available at Github and yours is one of the nicest. I am no coder at all (I can barely copy and paste code and pray for it to work. LoL). Your roadmap seems awesome. The only thing I really wish right now is the ability to ignore stocks and show only cryptos. But if I do that with the config file, I get an error.

By the way, have you seen the (https://fintic.io) project? I am surprised that there are no "full featured" matrix repositories here. I wish I could have more knowledge to code something like that.

Keep up the good work.

feram18, 2022-01-04:
Thank you for your kind words!

I honestly had not consider users who would only want to show cryptos and not stocks (or vice versa), my apologies. But I can surely make that work!

And yes, I actually saw that product being advertised on a few subreddits a while back. The main reason why I haven't considered adding other features to this project such as news, weather, etc. is because it's out of the scope of this project. Although, forex rates is something that can fall into the project's scope and I could add to this project in the future.

Thanks again for the feedback!

gcacciola, 2022-01-05:
Awesome thanks a lot for considering the crypto only feature.

Today I made a few improvements to my way simpler fork of the howchoo crypto-ticker. Involving layout, fonts and logos. learning a lot.

gcacciola, 2022-01-05:
Oh, as you said about news and weather, I wasn’t even considering them, what I really like about that project is the web interface to add simbols and customize some stuff in the fly.

feram18, 2022-01-05:
> Awesome thanks a lot for considering the crypto only feature.
> 
> Today I made a few improvements to my way simpler fork of the howchoo crypto-ticker. Involving layout, fonts and logos. learning a lot.

Awesome! Logos is something I can do as well, it just involves changing the layout quite a bit so that the image has a good enough resolution to look decent, so I'm considering adding this as an option later on.

feram18, 2022-01-05:
> Oh, as you said about news and weather, I wasn’t even considering them, what I really like about that project is the web interface to add simbols and customize some stuff in the fly.

Oh yeah, that's something I already have on my [roadmap](https://github.com/feram18/led-stock-ticker#roadmap)!

gcacciola, 2022-01-05:
Great, for me is a bit easier as I am working with a 128x32 panel, so I can play a lot with the layout.
I don't know if it helps you with something, but during my research I found these: 
https://github.com/coinwink/cryptocurrency-logos
https://cryptologos.cc

feram18, 2022-01-05:
> Great, for me is a bit easier as I am working with a 128x32 panel, so I can play a lot with the layout. I don't know if it helps you with something, but during my research I found these: https://github.com/coinwink/cryptocurrency-logos https://cryptologos.cc

Thanks! It could be useful. The yfinance library this project uses provides with logos for stocks but not cryptos. I'll take a look and see.

feram18, 2022-01-15:
Quick update on this, I've released v0.2.2 with the ability to pick cryptos or stocks only, as well as configurable date format, and update and rotation rate through the configuration script. Thank you all for the suggestions!

rj2764, 2022-12-29:
what is the least expensive Rasp pi I can use to make one of these?

j0el, 2024-03-15:
Great work!  Thanks for sharing.  I bought a 64x64 matrix for around $30, and it was easy to add the json for that dimension.
The install worked perfectly on a Pi3,  but I will try running it on a Pi zero when I get one as it will then fit into a small frame.