feram18/led-stock-ticker #6: ./config.py

TMARTKSU, 2022-03-21:
When I type this in the code runs. However I get a message saying: ModuleNotFoundError: No module names 'rgbmatrix'

please help:)

TMARTKSU, 2022-03-21:
Could this be because my RGB Matrix is not yet connected?
Also I am using a 16x32 matrix is it possible I can set it up to use on that?

feram18, 2022-03-26:
> When I type this in the code runs. However I get a message saying: ModuleNotFoundError: No module names 'rgbmatrix'
> 
> please help:)

If it can't find the `rgbmatrix` module, something must have gone wrong during the installation. Try re-running the `install.sh` script and check for any error messages.

feram18, 2022-03-26:
> Could this be because my RGB Matrix is not yet connected?
> Also I am using a 16x32 matrix is it possible I can set it up to use on that?

I believe the app would run whether the matrix is plugged in or not. I'm not sure if there is a way to know when it's connected, I'll have to look into that.

As for the 32x16: Unfortunately, I do not have support for that size at this time, since it would not be able to accommodate all the data with such little area to work with. The current minimum size is 62x32 (width x height). But perhaps a "lite" version could be worked out to show only the ticker and current price. If you're willing, you can play around with it by duplicating the existing `w62h32.json` layout file located in `/config/layout/` and rename it `w32h16.json` and change the coordinate values.

DJsupreme98, 2022-03-26:
Hi, I ran the project to and got the same error. I am running a adafruit 64x32 with a adfruti bonnet. Please could you assist

   ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/pi/led-stock-ticker/rpi-rgb-led-matrix/bindings/python/setup.py'"'"'; __file__='"'"'/home/pi/led-stock-ticker/rpi-rgb-led-matrix/bindings/python/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' develop --no-deps --user --prefix=
         cwd: /home/pi/led-stock-ticker/rpi-rgb-led-matrix/bindings/python/
    Complete output (154 lines):
    running develop
    running egg_info
    writing rgbmatrix.egg-info/PKG-INFO
    writing dependency_links to rgbmatrix.egg-info/dependency_links.txt
    writing top-level names to rgbmatrix.egg-info/top_level.txt
    reading manifest file 'rgbmatrix.egg-info/SOURCES.txt'
    writing manifest file 'rgbmatrix.egg-info/SOURCES.txt'
    running build_ext
    building 'graphics' extension
    arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I../../include -I/usr/include/python3.9 -c rgbmatrix/graphics.cpp -o build/temp.linux-armv7l-3.9/rgbmatrix/graphics.o -O3 -Wall
    rgbmatrix/graphics.cpp: In function ‘int __Pyx_modinit_type_init_code()’:
    rgbmatrix/graphics.cpp:3442:41: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
     3442 |   __pyx_type_9rgbmatrix_8graphics_Color.tp_print = 0;
          |                                         ^~~~~~~~
    rgbmatrix/graphics.cpp:3450:40: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
     3450 |   __pyx_type_9rgbmatrix_8graphics_Font.tp_print = 0;
          |                                        ^~~~~~~~
    In file included from /usr/include/python3.9/unicodeobject.h:1026,
                     from /usr/include/python3.9/Python.h:97,
                     from rgbmatrix/graphics.cpp:4:
    rgbmatrix/graphics.cpp: In function ‘int __Pyx_ParseOptionalKeywords(PyObject*, PyObject***, PyObject*, PyObject**, Py_ssize_t, const char*)’:
    /usr/include/python3.9/cpython/unicodeobject.h:451:75: warning: ‘Py_ssize_t _PyUnicode_get_wstr_length(PyObject*)’ is deprecated [-Wdeprecated-declarations]
      451 | #define PyUnicode_WSTR_LENGTH(op) _PyUnicode_get_wstr_length((PyObject*)op)
          |                                                                           ^
    /usr/include/python3.9/cpython/unicodeobject.h:261:7: note: in expansion of macro ‘PyUnicode_WSTR_LENGTH’
      261 |       PyUnicode_WSTR_LENGTH(op) :                    \
          |       ^~~~~~~~~~~~~~~~~~~~~
    rgbmatrix/graphics.cpp:3884:22: note: in expansion of macro ‘PyUnicode_GET_SIZE’
     3884 |                     (PyUnicode_GET_SIZE(**name) != PyUnicode_GET_SIZE(key)) ? 1 :
          |                      ^~~~~~~~~~~~~~~~~~
    /usr/include/python3.9/cpython/unicodeobject.h:446:26: note: declared here
      446 | static inline Py_ssize_t _PyUnicode_get_wstr_length(PyObject *op) {
          |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~
    /usr/include/python3.9/cpython/unicodeobject.h:262:52: warning: ‘Py_UNICODE* PyUnicode_AsUnicode(PyObject*)’ is deprecated [-Wdeprecated-declarations]
      262 |       ((void)PyUnicode_AsUnicode(_PyObject_CAST(op)),\
          |                                                    ^
    rgbmatrix/graphics.cpp:3884:22: note: in expansion of macro ‘PyUnicode_GET_SIZE’
     3884 |                     (PyUnicode_GET_SIZE(**name) != PyUnicode_GET_SIZE(key)) ? 1 :
          |                      ^~~~~~~~~~~~~~~~~~
    /usr/include/python3.9/cpython/unicodeobject.h:580:45: note: declared here
      580 | Py_DEPRECATED(3.3) PyAPI_FUNC(Py_UNICODE *) PyUnicode_AsUnicode(
          |                                             ^~~~~~~~~~~~~~~~~~~
    /usr/include/python3.9/cpython/unicodeobject.h:451:75: warning: ‘Py_ssize_t _PyUnicode_get_wstr_length(PyObject*)’ is deprecated [-Wdeprecated-declarations]
      451 | #define PyUnicode_WSTR_LENGTH(op) _PyUnicode_get_wstr_length((PyObject*)op)
          |                                                                           ^
    /usr/include/python3.9/cpython/unicodeobject.h:264:8: note: in expansion of macro ‘PyUnicode_WSTR_LENGTH’
      264 |        PyUnicode_WSTR_LENGTH(op)))
          |        ^~~~~~~~~~~~~~~~~~~~~
    rgbmatrix/graphics.cpp:3884:22: note: in expansion of macro ‘PyUnicode_GET_SIZE’
     3884 |                     (PyUnicode_GET_SIZE(**name) != PyUnicode_GET_SIZE(key)) ? 1 :
          |                      ^~~~~~~~~~~~~~~~~~
    /usr/include/python3.9/cpython/unicodeobject.h:446:26: note: declared here
      446 | static inline Py_ssize_t _PyUnicode_get_wstr_length(PyObject *op) {
          |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~
    /usr/include/python3.9/cpython/unicodeobject.h:451:75: warning: ‘Py_ssize_t _PyUnicode_get_wstr_length(PyObject*)’ is deprecated [-Wdeprecated-declarations]
      451 | #define PyUnicode_WSTR_LENGTH(op) _PyUnicode_get_wstr_length((PyObject*)op)
[... cut ...]
ERROR: Command errored out with exit status 1: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/pi/led-stock-ticker/rpi-rgb-led-matrix/bindings/python/setup.py'"'"'; __file__='"'"'/home/pi/led-stock-ticker/rpi-rgb-led-matrix/bindings/python/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' develop --no-deps --user --prefix= Check the logs for full command output.

feram18, 2022-10-31:
Support for 32x16 added in [v0.3.1](https://github.com/feram18/led-stock-ticker/releases/tag/v0.3.1).

closed 2022-10-31 by feram18

DJsupreme98, 2022-11-16:
Hi Luis,

I seem to have an issue when installing the ./install.sh command.
Seem to always get stuck on numpy install then shutdown. I have tried multiple time and on different OS.

Please help.

Kind Regards

Antonio

DJsupreme98, 2022-11-17:
Any idea as to why the colors are not displaying correctly on the matrix. Blue seems to be green and red is more like a pink colour.

feram18, 2022-11-20:
> colors are not displaying correctly on the matrix. Blue seems to be green and red is more like a pink colour.

Have you tried running other software that uses the LED matrix, or the rgb-led-matrix library samples?

If the colors aren't showing properly there either, you can try playing around with the `--led-rgb-sequence` and `--led-row-addr-type` flags. I personally have a matrix that wasn't displaying colors properly out of the box and had to change the values for these properties.

If other software/the samples seem to display the colors correctly, can you please attach a picture of what it looks like?

DJsupreme98, 2022-11-20:
Hi,

Thanks for the reply.

I tried it on other tickers and the colours display correctly.
Please could you explain as to how the sequence works please.

Thanks

feram18, 2022-11-20:
If other software is displaying colors correctly I would think the matrix is not the problem, so the flags would not really make a difference. Though I'm not sure how the problem would be this software specifically, especially if they are using the same library to render to the matrix.

Perhaps it's the shades of the colors I am using on this project that are not showing well. Did you previously use this same matrix for this project and the colors displayed correctly? Or is this the first time you're using the matrix for this project?

What other software did you try it with?

DJsupreme98, 2022-11-20:
I used your previous versions before with no issues.

I have used various other ticker software with no issue.
I am using adafruit - hat if that helps

feram18, 2022-11-20:
Perhaps something gone wrong with the matrix library.

You can reinstall the matrix library only. From the `led-stock-ticker` directory:

```sh
cd rpi-rgb-led-matrix
make build-python PYTHON="$(command -v python3)"
sudo make install-python PYTHON="$(command -v python3)"
```

If that does not fix the issue, I think it could be due to the specific color values I am using, since I changed the shade in some of them in a recent update. Are you able to attach a picture of what it looks like?

DJsupreme98, 2022-11-20:
Hi Luis,

I tried that and still no luck. Please see attached screenshot of my LED

￼