feram18/led-stock-ticker #7: Some ideas

Zobrob, 2022-05-07:
Hey thanks a lot for this awesome project and the nice documentation.
Even indices work 👍 
If you should ever consider an alternative design, I would love to have the logos for Bitcoin and Ethereum (instead of the history).

feram18, 2022-05-09:
Thanks for the feedback, suggestions are always welcome!

I am working on some new features, slowly due to lack of time, but coming along. Adding logos in place of history chart is actually one of the things on the to-do list. I'll keep the daily winners/losers in mind for a future update.

Zobrob, 2022-05-09:
That's great to hear, thank you.

feram18, 2022-12-26:
Thank you for the feedback! I'll update the README to mark the HAT/Bonnet as recommended but optional components. Thanks for the support!

closed 2022-12-26 by feram18