feram18/led-stock-ticker #8: Unable to install and start

copyfile, 2022-09-28:
I have a completely fresh install of raspberry PI OS and then followed all the commands to install the stock ticker. But for some reason it is not starting/showing anything on the led matrix. I tried a test script from adafruit and that works, so the connection and led matrix are working. Is there something I'm doing wrong? See below all the commands and results:

`pi@raspberrypi:~ $ sudo apt-get update
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done             
pi@raspberrypi:~ $ sudo apt-get install git python-pip
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package python-pip is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  python3-pip

E: Package 'python-pip' has no installation candidate
pi@raspberrypi:~ $ git clone --recursive https://github.com/feram18/led-stock-ticker.git
Cloning into 'led-stock-ticker'...
remote: Enumerating objects: 1075, done.
remote: Counting objects: 100% (85/85), done.
remote: Compressing objects: 100% (45/45), done.
remote: Total 1075 (delta 54), reused 53 (delta 40), pack-reused 990
Receiving objects: 100% (1075/1075), 44.82 MiB | 4.88 MiB/s, done.
Resolving deltas: 100% (685/685), done.
Submodule 'rpi-rgb-led-matrix' (https://github.com/hzeller/rpi-rgb-led-matrix.git) registered for path 'rpi-rgb-led-matrix'
Cloning into '/home/pi/led-stock-ticker/rpi-rgb-led-matrix'...
remote: Enumerating objects: 4892, done.        
remote: Total 4892 (delta 0), reused 0 (delta 0), pack-reused 4892        
Receiving objects: 100% (4892/4892), 22.24 MiB | 5.58 MiB/s, done.
Resolving deltas: 100% (3383/3383), done.
Submodule path 'rpi-rgb-led-matrix': checked out 'a56338db0f003d5236f2ce98c73a591d64a70852'
pi@raspberrypi:~ $ cd led-stock-ticker
pi@raspberrypi:~/led-stock-ticker $ chmod +x install.sh
pi@raspberrypi:~/led-stock-ticker $ ./install.sh
___________________________________________________________________
   __   _______    ______           __     _______     __          
  / /  / __/ _ \  / __/ /____  ____/ /__  /_  __(_)___/ /_____ ____
 / /__/ _// // / _\ \/ __/ _ \/ __/  '_/   / / / / __/  '_/ -_) __/
/____/___/____/ /___/\__/\___/\__/_/\_\   /_/ /_/\__/_/\_\__/_/   
___________________________________________________________________

Updating system & installing Python 3
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-dev is already the newest version (3.9.2-3).
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 51 not upgraded.

Installing rpi-rgb-led-matrix library...
make -C ./lib
make[1]: Entering directory '/home/pi/led-stock-ticker/rpi-rgb-led-matrix/lib'
g++ -I../include -W -Wall -Wextra -Wno-unused-parameter -O3 -g -fPIC  -DDEFAULT_HARDWARE='"regular"' -fno-exceptions -std=c++11 -c -o gpio.o gpio.cc
g++ -I../include -W -Wall -Wextra -Wno-unused-parameter -O3 -g -fPIC  -DDEFAULT_HARDWARE='"regular"' -fno-exceptions -std=c++11 -c -o led-matrix.o led-matrix.cc
g++ -I../include -W -Wall -Wextra -Wno-unused-parameter -O3 -g -fPIC  -DDEFAULT_HARDWARE='"regular"' -fno-exceptions -std=c++11 -c -o options-initialize.o options-initialize.cc
g++ -I../include -W -Wall -Wextra -Wno-unused-parameter -O3 -g -fPIC  -DDEFAULT_HARDWARE='"regular"' -fno-exceptions -std=c++11 -c -o framebuffer.o framebuffer.cc
g++ -I../include -W -Wall -Wextra -Wno-unused-parameter -O3 -g -fPIC  -DDEFAULT_HARDWARE='"regular"' -fno-exceptions -std=c++11 -c -o thread.o thread.cc
g++ -I../include -W -Wall -Wextra -Wno-unused-parameter -O3 -g -fPIC  -DDEFAULT_HARDWARE='"regular"' -fno-exceptions -std=c++11 -c -o bdf-font.o bdf-font.cc
g++ -I../include -W -Wall -Wextra -Wno-unused-parameter -O3 -g -fPIC  -DDEFAULT_HARDWARE='"regular"' -fno-exceptions -std=c++11 -c -o graphics.o graphics.cc
g++ -I../include -W -Wall -Wextra -Wno-unused-parameter -O3 -g -fPIC  -DDEFAULT_HARDWARE='"regular"' -fno-exceptions -std=c++11 -c -o led-matrix-c.o led-matrix-c.cc
cc  -I../include -W -Wall -Wextra -Wno-unused-parameter -O3 -g -fPIC  -DDEFAULT_HARDWARE='"regular"' -c -o hardware-mapping.o hardware-mapping.c
g++ -I../include -W -Wall -Wextra -Wno-unused-parameter -O3 -g -fPIC  -DDEFAULT_HARDWARE='"regular"' -fno-exceptions -std=c++11 -c -o pixel-mapper.o pixel-mapper.cc
g++ -I../include -W -Wall -Wextra -Wno-unused-parameter -O3 -g -fPIC  -DDEFAULT_HARDWARE='"regular"' -fno-exceptions -std=c++11 -c -o multiplex-mappers.o multiplex-mappers.cc
g++ -I../include -W -Wall -Wextra -Wno-unused-parameter -O3 -g -fPIC  -DDEFAULT_HARDWARE='"regular"' -fno-exceptions -std=c++11 -c -o content-streamer.o content-streamer.cc
[... 186 lines cut ...]
ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 48 from C header, got 40 from PyObject
pi@raspberrypi:~/led-stock-ticker $ sudo python3 main.py --led-gpio-mapping="adafruit-hat" --led-slowdown-gpio=2
Traceback (most recent call last):
  File "/home/pi/led-stock-ticker/main.py", line 6, in <module>
    from util.utils import args, led_matrix_options
  File "/home/pi/led-stock-ticker/util/utils.py", line 18, in <module>
    from util.holiday_calendar import MarketHolidayCalendar
  File "/home/pi/led-stock-ticker/util/holiday_calendar.py", line 1, in <module>
    from pandas.tseries.offsets import WeekOfMonth
  File "/usr/local/lib/python3.9/dist-packages/pandas/__init__.py", line 22, in <module>
    from pandas.compat import is_numpy_dev as _is_numpy_dev
  File "/usr/local/lib/python3.9/dist-packages/pandas/compat/__init__.py", line 15, in <module>
    from pandas.compat.numpy import (
  File "/usr/local/lib/python3.9/dist-packages/pandas/compat/numpy/__init__.py", line 4, in <module>
    from pandas.util.version import Version
  File "/usr/local/lib/python3.9/dist-packages/pandas/util/__init__.py", line 1, in <module>
    from pandas.util._decorators import (  # noqa:F401
  File "/usr/local/lib/python3.9/dist-packages/pandas/util/_decorators.py", line 14, in <module>
    from pandas._libs.properties import cache_readonly  # noqa:F401
  File "/usr/local/lib/python3.9/dist-packages/pandas/_libs/__init__.py", line 13, in <module>
    from pandas._libs.interval import Interval
  File "pandas/_libs/interval.pyx", line 1, in init pandas._libs.interval
ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 48 from C header, got 40 from PyObject

`

feram18, 2022-10-10:
> Package python-pip is not available, but is referred to by another package.
> This may mean that the package is missing, has been obsoleted, or
> is only available from another source
> However the following packages replace it:
> python3-pip
> 
> E: Package 'python-pip' has no installation candidate

I'll be updating the README to account for this mistake. Try `sudo apt-get install python3-pip -y` instead. Then try re-running the `update.sh` script.

---

> ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 48 from C header, got 40 from PyObject

This error appears to be related to the numpy dependency only, try reinstalling with:

`pip3 install --upgrade numpy`

copyfile, 2022-10-11:
Thanks for your reply! I have run both changes but unfortunatly it is not (yet) working. See below what's happening when trying to perform these actions:

pi@raspberrypi:~ $ sudo apt-get install python3-pip -y
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-pip is already the newest version (20.3.4-4+rpt1+deb11u1).
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 23 not upgraded.

pi@raspberrypi:~/led-stock-ticker $ pip3 install --upgrade numpy
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: numpy in /usr/local/lib/python3.9/dist-packages (1.23.3)
Collecting numpy
  Downloading https://www.piwheels.org/simple/numpy/numpy-1.23.3-cp39-cp39-linux_armv7l.whl (12.4 MB)
     |████████████████████████████████| 12.4 MB 72 kB/s 
  Downloading https://www.piwheels.org/simple/numpy/numpy-1.23.2-cp39-cp39-linux_armv7l.whl (12.3 MB)
     |████████████████████████████████| 12.3 MB 75 kB/s 

pi@raspberrypi:~/led-stock-ticker $ ./config.py
Traceback (most recent call last):
  File "/home/pi/led-stock-ticker/./config.py", line 10, in <module>
    from util.utils import read_json, write_json
  File "/home/pi/led-stock-ticker/util/utils.py", line 18, in <module>
    from util.holiday_calendar import MarketHolidayCalendar
  File "/home/pi/led-stock-ticker/util/holiday_calendar.py", line 1, in <module>
    from pandas.tseries.offsets import WeekOfMonth
  File "/usr/local/lib/python3.9/dist-packages/pandas/__init__.py", line 16, in <module>
    raise ImportError(
ImportError: Unable to import required dependencies:
numpy: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.9 from "/usr/bin/python3"
  * The NumPy version is: "1.23.3"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: libcblas.so.3: cannot open shared object file: No such file or directory

pi@raspberrypi:~/led-stock-ticker $ sudo python3 main.py --led-gpio-mapping="adafruit-hat" --led-slowdown-gpio=2
Traceback (most recent call last):
  File "/home/pi/led-stock-ticker/main.py", line 6, in <module>
    from util.utils import args, led_matrix_options
  File "/home/pi/led-stock-ticker/util/utils.py", line 18, in <module>
    from util.holiday_calendar import MarketHolidayCalendar
  File "/home/pi/led-stock-ticker/util/holiday_calendar.py", line 1, in <module>
    from pandas.tseries.offsets import WeekOfMonth
  File "/usr/local/lib/python3.9/dist-packages/pandas/__init__.py", line 16, in <module>
    raise ImportError(
ImportError: Unable to import required dependencies:
numpy: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.9 from "/usr/bin/python3"
  * The NumPy version is: "1.23.3"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: libcblas.so.3: cannot open shared object file: No such file or directory

pi@raspberrypi:~/led-stock-ticker $

copyfile, 2022-10-11:
I was able to get it working now with installing a legacy version of Rasbarian! Thanks!!

closed 2022-10-12 by feram18