feram18/led-stock-ticker #10: This is just amazing

renjithstalin, 2022-10-19:
This is pretty amazing, I have tried various tickers using max7219/epaper etc, and this is by far the best. Thank You so very much.
setup was a breeze and it works perfectly.
Hope I'm not asking too much here, but is it possible to add another panel, with a scrolling effect just like the ones in stock exchanges?

Few observations (maybe something I have to configure, didn't dig into the settings too much)
-P10 panel displays only  a portion of the data 
-ticker name scrolling is a bit jerky

Thanks again.

feram18, 2022-10-19:
Thank you for trying out this project!

> is it possible to add another panel, with a scrolling effect just like the ones in stock exchanges?

I'm not sure what you mean. Do you mean like a screen to scroll through some custom text? Or like scrolling through all the tickers together?

> P10 panel displays only a portion of the data

Are you able to attach a picture of this issue? It would help to get a better idea if it's an issue that should be addressed in the code, or if could be fixed on your end through some configuration.

> ticker name scrolling is a bit jerky

I've noticed some issues with scrolling myself, I am working on a fix for the next release.

renjithstalin, 2022-10-20:
Thanks for the response, P10 panels displays like this  
[image]
[image]
[image]

P5 panels works perfectly.
[image]

_I'm not sure what you mean. Do you mean like a screen to scroll through some custom text? Or like scrolling through all the tickers together?_
I was wondering if multiple panels can be daisy-chained, to create a long one (at least with 2) to make a scrolling display, something like this : 
[image]

Thanks again.

feram18, 2022-10-31:
Added 32x16 layout support in [v0.3.1](https://github.com/feram18/led-stock-ticker/releases/tag/v0.3.1).

closed 2022-10-31 by feram18