feram18/led-stock-ticker #12: Bug with chain

zamzyy, 2022-11-19:
I'm having some trouble getting it to display to both of the matrix displays using --led-chain=2, the code will run but not display anything and then just force quit itself. Any Ideas or am I doing something wrong?

please and thank you, your stock ticker is exactly what I would want it displayed and it is fantastic, great job.

closed 2022-11-19 by zamzyy

feram18, 2022-11-20:
> trouble getting it to display to both of the matrix displays using --led-chain=2

For reference to anyone else who may be having issues: If your boards are chained you can simply set the values for the `--led-rows` and `--led-cols` flags to the values they amount to together. For example. two 32x32 (width x height) boards would add up to 64x32 and the flags would be set to `--led-cols=64` and `--led-rows=32`.

zamzyy, 2022-11-20:
First I wanted to thank you for the quick reply,

But I was able to figure it out I tried using --led-cols command but it
still wouldn't work. I found that the 128x32.Json file in the utils folder
was missing so I copied the 64x32 one and renamed it and it worked! Thanks
for your help!

feram18, 2022-11-20:
I see. Though currently I don't have support for the 128x32, your solution works. Official support for this board size is on the roadmap for the next release. I'll also add an error message to be logged when this unsupported board size situation occurs to clear things up.