feram18/led-stock-ticker #13: 'Stock' object has no attribute 'name

BraytonL, 2022-12-19:
> `12/19/2022 05:47:11 AM ERROR: No data available for TSLA.
12/19/2022 05:47:11 AM ERROR: 'Stock' object has no attribute 'name'
Traceback (most recent call last):
  File "main.py", line 48, in <module>
    main()
  File "main.py", line 22, in main
    data = Data(config)
  File "<string>", line 9, in __init__
  File "/home/pi/led-stock-ticker/api/data.py", line 38, in __post_init__
    self.initialize()
  File "/home/pi/led-stock-ticker/api/data.py", line 50, in initialize
    self.fetch_stock(stock, self.config.currency)
  File "/usr/local/lib/python3.7/dist-packages/multitasking/__init__.py", line 109, in async_method
    return callee(*args, **kwargs)
  File "/home/pi/led-stock-ticker/api/data.py", line 91, in fetch_stock
    stock = Stock(symbol, currency)
  File "<string>", line 9, in __init__
  File "/home/pi/led-stock-ticker/data/ticker.py", line 31, in __post_init__
    self.initialize()
  File "/home/pi/led-stock-ticker/data/stock.py", line 14, in initialize
    self.name = self.name\
AttributeError: 'Stock' object has no attribute 'name'
12/19/2022 05:50:24 AM ERROR: No data available for TSLA.
12/19/2022 05:50:24 AM ERROR: 'Stock' object has no attribute 'name'
Traceback (most recent call last):
  File "main.py", line 48, in <module>
    main()
  File "main.py", line 22, in main
    data = Data(config)
  File "<string>", line 9, in __init__
  File "/home/pi/led-stock-ticker/api/data.py", line 38, in __post_init__
    self.initialize()
  File "/home/pi/led-stock-ticker/api/data.py", line 50, in initialize
    self.fetch_stock(stock, self.config.currency)
  File "/usr/local/lib/python3.7/dist-packages/multitasking/__init__.py", line 109, in async_method
    return callee(*args, **kwargs)
  File "/home/pi/led-stock-ticker/api/data.py", line 91, in fetch_stock
    stock = Stock(symbol, currency)
  File "<string>", line 9, in __init__
  File "/home/pi/led-stock-ticker/data/ticker.py", line 31, in __post_init__
    self.initialize()
  File "/home/pi/led-stock-ticker/data/stock.py", line 14, in initialize
    self.name = self.name\
AttributeError: 'Stock' object has no attribute 'name'
`

copyfile, 2022-12-20:
I have exactly the same problem since a couple of days. Tried a complete re-install but still the same issue:

pi@raspberrypi:~/led-stock-ticker $ sudo python3 main.py --led-gpio-mapping="adafruit-hat" --led-slowdown-gpio=2
📊 LED-Stock-Ticker - v0.3.3 (64x32)
Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.7/dist-packages/multitasking/__init__.py", line 104, in _run_via_pool
    return callee(*args, **kwargs)
  File "/home/pi/led-stock-ticker/api/data.py", line 105, in fetch_crypto
    crypto = Crypto(symbol, currency)
  File "<string>", line 9, in __init__
  File "/home/pi/led-stock-ticker/data/ticker.py", line 31, in __post_init__
    self.initialize()
  File "/home/pi/led-stock-ticker/data/crypto.py", line 18, in initialize
    self.name = self.name.replace(' USD', '')
AttributeError: 'Crypto' object has no attribute 'name'

Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.7/dist-packages/multitasking/__init__.py", line 104, in _run_via_pool
    return callee(*args, **kwargs)
  File "/home/pi/led-stock-ticker/api/data.py", line 91, in fetch_stock
    stock = Stock(symbol, currency)
  File "<string>", line 9, in __init__
  File "/home/pi/led-stock-ticker/data/ticker.py", line 31, in __post_init__
    self.initialize()
  File "/home/pi/led-stock-ticker/data/stock.py", line 14, in initialize
    self.name = self.name\
AttributeError: 'Stock' object has no attribute 'name'

Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.7/dist-packages/multitasking/__init__.py", line 104, in _run_via_pool
    return callee(*args, **kwargs)
  File "/home/pi/led-stock-ticker/api/data.py", line 91, in fetch_stock
    stock = Stock(symbol, currency)
  File "<string>", line 9, in __init__
  File "/home/pi/led-stock-ticker/data/ticker.py", line 31, in __post_init__
    self.initialize()
  File "/home/pi/led-stock-ticker/data/stock.py", line 14, in initialize
    self.name = self.name\
AttributeError: 'Stock' object has no attribute 'name'

Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.7/dist-packages/multitasking/__init__.py", line 104, in _run_via_pool
    return callee(*args, **kwargs)
  File "/home/pi/led-stock-ticker/api/data.py", line 91, in fetch_stock
    stock = Stock(symbol, currency)
  File "<string>", line 9, in __init__
  File "/home/pi/led-stock-ticker/data/ticker.py", line 31, in __post_init__
    self.initialize()
  File "/home/pi/led-stock-ticker/data/stock.py", line 14, in initialize
    self.name = self.name\
AttributeError: 'Stock' object has no attribute 'name'

Exception in thread Thread-6:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.7/dist-packages/multitasking/__init__.py", line 104, in _run_via_pool
    return callee(*args, **kwargs)
  File "/home/pi/led-stock-ticker/api/data.py", line 105, in fetch_crypto
    crypto = Crypto(symbol, currency)
  File "<string>", line 9, in __init__
  File "/home/pi/led-stock-ticker/data/ticker.py", line 31, in __post_init__
    self.initialize()
  File "/home/pi/led-stock-ticker/data/crypto.py", line 18, in initialize
    self.name = self.name.replace(' USD', '')
AttributeError: 'Crypto' object has no attribute 'name'

Exception in thread Thread-5:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.7/dist-packages/multitasking/__init__.py", line 104, in _run_via_pool
    return callee(*args, **kwargs)
  File "/home/pi/led-stock-ticker/api/data.py", line 105, in fetch_crypto
    crypto = Crypto(symbol, currency)
  File "<string>", line 9, in __init__
  File "/home/pi/led-stock-ticker/data/ticker.py", line 31, in __post_init__
    self.initialize()
  File "/home/pi/led-stock-ticker/data/crypto.py", line 18, in initialize
    self.name = self.name.replace(' USD', '')
AttributeError: 'Crypto' object has no attribute 'name'

feram18, 2022-12-21:
I experienced this issue myself a while back but didn't see it again until yesterday. I'll begin working on a fix.

BraytonL, 2022-12-21:
@feram18 Thank you! Any idea what it is related to considering your past experience?

feram18, 2022-12-21:
It looks like something broke on yfinance version 0.1.74. After upgrading to the latest release, the issue is not coming up anymore. Testing a bit further and fixing a couple other things before I push a new release.

BraytonL, 2022-12-23:
@feram18 I know you haven't officially released a bug-fix yet, but even trying the code in the dev branch I am encountering the same `'Stock' object has no attribute 'name'` error. 

I also tried forking the repository and merging dev -> master, just to see what the changes really were. It seems like the update to YFinance did not fix the issue- unless I am missing something important.

Thank you for all your hard work you put into this project. It is amazing and I hope this issue can be resolved!

feram18, 2022-12-23:
Did you upgrade the yfinance version on your environment? If not, try `pip3 install yfinance` and that should install the latest version. I haven't seen the error again. If I go back to yfinance version 0.1.74, I see the error again so I'm confident the error is related to the older yfinance version.

Holychris56, 2022-12-23:
Which version of yfinance are you running? I am on the latest, with the release of [0.1.90] and I am still getting the same error.

Holychris56, 2022-12-23:
figured it out. do sudo pip3 install yfinance --upgrade.

copyfile, 2022-12-24:
Is there a way for me to already try the dev version? Not sure how that works installing wise. 

In the meantime I tried the Yfinance update but then I get a lot of error's (not sure if this step will me mandatory/needed for the dev/updated version):

`build_py.py:202: SetuptoolsDeprecationWarning:     Installing 'cryptography.hazmat.bindings._rust' as data is deprecated, please list it in `packages`.
        !!
    
    
        ############################
        # Package would be ignored #
        ############################
        Python recognizes 'cryptography.hazmat.bindings._rust' as an importable package,
        but it is not listed in the `packages` configuration of setuptools.
    
        'cryptography.hazmat.bindings._rust' has been automatically added to the distribution only
        because it may contain data files, but this behavior is likely to change
        in future versions of setuptools (and therefore is considered deprecated).
    
        Please make sure that 'cryptography.hazmat.bindings._rust' is included as a package by using
        the `packages` configuration field or the proper discovery methods
        (for example by using `find_namespace_packages(...)`/`find_namespace:`
        instead of `find_packages(...)`/`find:`).
    
        You can read more about "package discovery" and "data files" on setuptools
        documentation page.
    
    
    !!
    
      check.warn(importable)
    copying src/cryptography/py.typed -> build/lib.linux-armv7l-cpython-37/cryptography
    copying src/cryptography/hazmat/bindings/_openssl.pyi -> build/lib.linux-armv7l-cpython-37/cryptography/hazmat/bindings
    creating build/lib.linux-armv7l-cpython-37/cryptography/hazmat/bindings/_rust
    copying src/cryptography/hazmat/bindings/_rust/__init__.pyi -> build/lib.linux-armv7l-cpython-37/cryptography/hazmat/bindings/_rust
    copying src/cryptography/hazmat/bindings/_rust/asn1.pyi -> build/lib.linux-armv7l-cpython-37/cryptography/hazmat/bindings/_rust
    copying src/cryptography/hazmat/bindings/_rust/ocsp.pyi -> build/lib.linux-armv7l-cpython-37/cryptography/hazmat/bindings/_rust
    copying src/cryptography/hazmat/bindings/_rust/x509.pyi -> build/lib.linux-armv7l-cpython-37/cryptography/hazmat/bindings/_rust
    running build_ext
    running build_rust
    
        =============================DEBUG ASSISTANCE=============================
        If you are seeing a compilation error please try the following steps to
        successfully install cryptography:
        1) Upgrade to the latest pip and try again. This will fix errors for most
           users. See: https://pip.pypa.io/en/stable/installing/#upgrading-pip
        2) Read https://cryptography.io/en/latest/installation/ for specific
           instructions for your platform.
        3) Check our frequently asked questions for more information:
           https://cryptography.io/en/latest/faq/
        4) Ensure you have a recent Rust toolchain installed:
           https://cryptography.io/en/latest/installation/#rust
    
        Python: 3.7.3
        platform: Linux-5.10.103-v7+-armv7l-with-debian-10.13
        pip: 18.1
        setuptools: 65.6.3
        setuptools_rust: 1.5.2
        rustc: n/a
        =============================DEBUG ASSISTANCE=============================
    
    error: can't find Rust compiler
    
    If you are using an outdated pip version, it is possible a prebuilt wheel is available for this package but pip is not able to install from it. Installing from the wheel would avoid the need for a Rust compiler.
    
    To update pip, run:
    
        pip install --upgrade pip
    
    and then retry package installation.
    
    If you did intend to build this package from source, try installing a Rust compiler from your system package manager and ensure it is on the PATH during installation. Alternatively, rustup (available at https://rustup.rs) is the recommended way to download and update the Rust compiler toolchain.
    
    This package requires Rust >=1.48.0.
    
    ----------------------------------------
  Can't roll back cryptography; was not uninstalled
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-jcwn3lna/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-yxklrpe_/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-jcwn3lna/cryptography/
`

feram18, 2022-12-24:
> Is there a way for me to already try the dev version? Not sure how that works installing wise.

```sh
git fetch --all
git checkout origin/dev
sudo pip3 install -r requirements.txt
```

You might have to do a `git reset --hard` or `git stash` before checking out the `origin/dev` branch.

copyfile, 2022-12-24:
Thanks this worked for changing to the dev version. Only when running the install I get a similar error message as in my previous message. Any idea how I can fix this?

pi@raspberrypi:~/led-stock-ticker $ sudo pip3 install -r requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: jsonschema~=4.16.0 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 1)) (4.16.0)
Requirement already satisfied: lxml>=4.5.1 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 2)) (4.9.1)
Requirement already satisfied: numpy>=1.16.6 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 3)) (1.21.6)
Requirement already satisfied: multitasking~=0.0.9 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 4)) (0.0.11)
Requirement already satisfied: pandas>=0.23.1 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 5)) (1.3.5)
Collecting pillow==9.3.0 (from -r requirements.txt (line 6))
  Downloading https://www.piwheels.org/simple/pillow/Pillow-9.3.0-cp37-cp37m-linux_armv7l.whl (812kB)
    100% |████████████████████████████████| 819kB 319kB/s 
Requirement already satisfied: pytz>=2021.1 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 7)) (2022.6)
Requirement already satisfied: questionary>=1.10.0 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 8)) (1.10.0)
Requirement already satisfied: requests>=2.22.0 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 9)) (2.28.1)
Requirement already satisfied: wheel>=0.37.0 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 10)) (0.38.4)
Collecting yfinance==0.2.3 (from -r requirements.txt (line 11))
  Using cached https://files.pythonhosted.org/packages/15/49/9cf3dfe15e9389d2a0f701b0a273bbc8c5f18183541c3abcc89f6d3e0445/yfinance-0.2.3-py2.py3-none-any.whl
Requirement already satisfied: attrs>=17.4.0 in /usr/local/lib/python3.7/dist-packages (from jsonschema~=4.16.0->-r requirements.txt (line 1)) (22.1.0)
Requirement already satisfied: typing-extensions; python_version < "3.8" in /usr/local/lib/python3.7/dist-packages (from jsonschema~=4.16.0->-r requirements.txt (line 1)) (4.4.0)
Requirement already satisfied: importlib-resources>=1.4.0; python_version < "3.9" in /usr/local/lib/python3.7/dist-packages (from jsonschema~=4.16.0->-r requirements.txt (line 1)) (5.10.0)
Requirement already satisfied: importlib-metadata; python_version < "3.8" in /usr/local/lib/python3.7/dist-packages (from jsonschema~=4.16.0->-r requirements.txt (line 1)) (5.0.0)
Requirement already satisfied: pkgutil-resolve-name>=1.3.10; python_version < "3.9" in /usr/local/lib/python3.7/dist-packages (from jsonschema~=4.16.0->-r requirements.txt (line 1)) (1.3.10)
Requirement already satisfied: pyrsistent!=0.17.0,!=0.17.1,!=0.17.2,>=0.14.0 in /usr/local/lib/python3.7/dist-packages (from jsonschema~=4.16.0->-r requirements.txt (line 1)) (0.19.2)
Requirement already satisfied: python-dateutil>=2.7.3 in /usr/local/lib/python3.7/dist-packages (from pandas>=0.23.1->-r requirements.txt (line 5)) (2.8.2)
Requirement already satisfied: prompt_toolkit<4.0,>=2.0 in /usr/local/lib/python3.7/dist-packages (from questionary>=1.10.0->-r requirements.txt (line 8)) (3.0.32)
Requirement already satisfied: idna<4,>=2.5 in /usr/lib/python3/dist-packages (from requests>=2.22.0->-r requirements.txt (line 9)) (2.6)
Requirement already satisfied: certifi>=2017.4.17 in /usr/lib/python3/dist-packages (from requests>=2.22.0->-r requirements.txt (line 9)) (2018.8.24)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in /usr/lib/python3/dist-packages (from requests>=2.22.0->-r requirements.txt (line 9)) (1.24.1)
Requirement already satisfied: charset-normalizer<3,>=2 in /usr/local/lib/python3.7/dist-packages (from requests>=2.22.0->-r requirements.txt (line 9)) (2.1.1)
Collecting cryptography>=3.3.2 (from yfinance==0.2.3->-r requirements.txt (line 11))
  Using cached https://files.pythonhosted.org/packages/e3/3f/41186b1f2fd86a542d399175f6b8e43f82cd4dfa51235a0b030a042b811a/cryptography-38.0.4.tar.gz
  Installing build dependencies ... done
Requirement already satisfied: appdirs>=1.4.4 in /usr/local/lib/python3.7/dist-packages (from yfinance==0.2.3->-r requirements.txt (line 11)) (1.4.4)
Requirement already satisfied: beautifulsoup4>=4.11.1 in /usr/local/lib/python3.7/dist-packages (from yfinance==0.2.3->-r requirements.txt (line 11)) (4.11.1)
Requirement already satisfied: frozendict>=2.3.4 in /usr/local/lib/python3.7/dist-packages (from yfinance==0.2.3->-r requirements.txt (line 11)) (2.3.4)
Requirement already satisfied: html5lib>=1.1 in /usr/local/lib/python3.7/dist-packages (from yfinance==0.2.3->-r requirements.txt (line 11)) (1.1)
Requirement already satisfied: zipp>=3.1.0; python_version < "3.10" in /usr/local/lib/python3.7/dist-packages (from importlib-resources>=1.4.0; python_version < "3.9"->jsonschema~=4.16.0->-r requirements.txt (line 1)) (3.10.0)
Requirement already satisfied: six>=1.5 in /usr/lib/python3/dist-packages (from python-dateutil>=2.7.3->pandas>=0.23.1->-r requirements.txt (line 5)) (1.12.0)
Requirement already satisfied: wcwidth in /usr/local/lib/python3.7/dist-packages (from prompt_toolkit<4.0,>=2.0->questionary>=1.10.0->-r requirements.txt (line 8)) (0.2.5)
[... cut ...]
  Can't roll back cryptography; was not uninstalled
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-lvopue4p/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-4gnlwu7n/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-lvopue4p/cryptography/
pi@raspberrypi:~/led-stock-ticker $

feram18, 2022-12-26:
It seems the latest version of `yfinance` now uses `cryptography`, and needs version `>= 3.3.2` of this package.

Going off `cryptography`'s [FAQs](https://cryptography.io/en/latest/faq/), it seems you just need to update your pip version.
You're currently at `18.1`, and the latest version is `22.3.1`. Update using `pip3 install --upgrade pip`.

feram18, 2022-12-26:
[v0.3.4](https://github.com/feram18/led-stock-ticker/releases/tag/v0.3.4) should take care of the "'Stock' object has no attribute 'name'" issue.

copyfile, 2022-12-26:
> It seems the latest version of `yfinance` now uses `cryptography`, and needs version `>= 3.3.2` of this package.
> 
> Going off `cryptography`'s [FAQs](https://cryptography.io/en/latest/faq/), it seems you just need to update your pip version.
> 
> > Building cryptography from source requires you have [Rust installed and available](https://cryptography.io/en/latest/installation/#rust) on your PATH. You may be able to fix this by upgrading to a newer version of pip which will install a pre-compiled cryptography wheel. If not, you’ll need to install Rust. Follow the [instructions](https://cryptography.io/en/latest/installation/#rust) to ensure you install a recent Rust version.
> 
> You're currently at `18.1`, and the latest version is `22.3.1`. Update using `pip3 install --upgrade pip`.

Yeah that's what I thought as well, but then I get the below message so not sure what's happening. Any idea how to fix this?

pi@raspberrypi:~ $ pip3 install --upgrade pip
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pip in ./.local/lib/python3.7/site-packages (22.3.1)

copyfile, 2022-12-27:
Looks to be in my RPI install. Just did a fresh install and now it is working perfectly again, thanks!

closed 2023-01-03 by feram18