feram18/led-stock-ticker #15: Percentage not updated

copyfile, 2023-01-11:
I'm running the latest version of the stock ticker now and I noticed that the percentage change is not updated. Once I do a restart of me RPI it is working again for that moment, it shows the correct percentage. I created to cronjob to restart the RPI a couple of times a day but this didn't really solve the issue, only for right after the restart. The stockprice itself is correct, so it is only limited to the percentage. Any idea what this is, and what I can/need to do to solve this?

feram18, 2023-01-14:
Are you seeing the percentage updated at all throughout the day, or did it stay at the same value it had when it first ran?

copyfile, 2023-01-18:
Have tested it a couple of days now but it looks to be okay now. Thanks!

closed 2023-01-18 by copyfile