feram18/led-stock-ticker #16: object has no attribute 'name'

copyfile, 2023-01-13:
Since yesterday I have the following error (again). I think it is exactly the same one as a couple of weeks ago. Any idea how to solve this now?

Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.9/dist-packages/multitasking/__init__.py", line 104, in _run_via_pool
    return callee(*args, **kwargs)
  File "/home/pi/led-stock-ticker/api/data.py", line 91, in fetch_stock
    stock = Stock(symbol, currency)
  File "<string>", line 10, in __init__
  File "/home/pi/led-stock-ticker/data/ticker.py", line 31, in __post_init__
    self.initialize()
  File "/home/pi/led-stock-ticker/data/stock.py", line 13, in initialize
    self.logo_url = self.yf_ticker.info.get('logo_url', None)
AttributeError: 'NoneType' object has no attribute 'get'
Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.9/dist-packages/multitasking/__init__.py", line 104, in _run_via_pool
    return callee(*args, **kwargs)
  File "/home/pi/led-stock-ticker/api/data.py", line 91, in fetch_stock
    stock = Stock(symbol, currency)
  File "<string>", line 10, in __init__
  File "/home/pi/led-stock-ticker/data/ticker.py", line 31, in __post_init__
    self.initialize()
  File "/home/pi/led-stock-ticker/data/stock.py", line 13, in initialize
    self.logo_url = self.yf_ticker.info.get('logo_url', None)
AttributeError: 'NoneType' object has no attribute 'get'
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.9/dist-packages/multitasking/__init__.py", line 104, in _run_via_pool
    return callee(*args, **kwargs)
  File "/home/pi/led-stock-ticker/api/data.py", line 91, in fetch_stock
    stock = Stock(symbol, currency)
  File "<string>", line 10, in __init__
  File "/home/pi/led-stock-ticker/data/ticker.py", line 31, in __post_init__
    self.initialize()
  File "/home/pi/led-stock-ticker/data/stock.py", line 13, in initialize
    self.logo_url = self.yf_ticker.info.get('logo_url', None)
AttributeError: 'NoneType' object has no attribute 'get'
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.9/dist-packages/multitasking/__init__.py", line 104, in _run_via_pool
    return callee(*args, **kwargs)
  File "/home/pi/led-stock-ticker/api/data.py", line 91, in fetch_stock
    stock = Stock(symbol, currency)
  File "<string>", line 10, in __init__
  File "/home/pi/led-stock-ticker/data/ticker.py", line 31, in __post_init__
    self.initialize()
  File "/home/pi/led-stock-ticker/data/stock.py", line 13, in initialize
    self.logo_url = self.yf_ticker.info.get('logo_url', None)
AttributeError: 'NoneType' object has no attribute 'get'
Exception in thread Thread-5:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
[... 467 lines cut ...]
    return callee(*args, **kwargs)
  File "/home/pi/led-stock-ticker/api/data.py", line 105, in fetch_crypto
    crypto = Crypto(symbol, currency)
  File "<string>", line 10, in __init__
  File "/home/pi/led-stock-ticker/data/ticker.py", line 31, in __post_init__
    self.initialize()
  File "/home/pi/led-stock-ticker/data/crypto.py", line 18, in initialize
    self.name = self.name.replace(' USD', '')
AttributeError: 'Crypto' object has no attribute 'name'
Exception in thread Thread-35:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.9/dist-packages/multitasking/__init__.py", line 104, in _run_via_pool
    return callee(*args, **kwargs)
  File "/home/pi/led-stock-ticker/api/data.py", line 105, in fetch_crypto
    crypto = Crypto(symbol, currency)
  File "<string>", line 10, in __init__
  File "/home/pi/led-stock-ticker/data/ticker.py", line 31, in __post_init__
    self.initialize()
  File "/home/pi/led-stock-ticker/data/crypto.py", line 18, in initialize
    self.name = self.name.replace(' USD', '')
AttributeError: 'Crypto' object has no attribute 'name'

feram18, 2023-01-14:
Investigating the issue. Apologies for the inconvenience.

feram18, 2023-01-14:
It appears to be an error with `yfinance`'s data scrapper. This [issue](https://github.com/ranaroussi/yfinance/issues/1291) has some information on what's going on if you're interested in learning more. They seem to have a solution and will be making a new release with a fix soon. I will update the dependency version and push a new release as soon as it's up.

feram18, 2023-01-15:
[v0.3.5](https://github.com/feram18/led-stock-ticker/releases/tag/v0.3.5) upgrades the `yfinance` dependency to resolve this issue. Be aware this dependency has been a bit unstable lately due to changes in Yahoo Finance's site.

closed 2023-01-15 by feram18