feram18/led-stock-ticker #17: hello i did fresh install. it not works actualy

zhilly, 2023-01-28:
hello,

it not  work at me since last update.

sudo python3 main.py --led-gpio-mapping="adafruit-ha                                                                                                             t" --led-slowdown-gpio=2
Suggestion: to slightly improve display update, add
        isolcpus=3
at the end of /boot/cmdline.txt and reboot (see README.md)
📊 LED-Stock-Ticker - v0.3.6 (64x32)
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.9/dist-packages/multitasking/__init__.py", line 1                                                                                                             04, in _run_via_pool
    return callee(*args, **kwargs)
  File "/home/pi/led-stock-ticker/api/data.py", line 91, in fetch_stock
    stock = Stock(symbol, currency)
  File "<string>", line 10, in __init__
  File "/home/pi/led-stock-ticker/data/ticker.py", line 31, in __post_init__
    self.initialize()
  File "/home/pi/led-stock-ticker/data/stock.py", line 12, in initialize
    super(Stock, self).initialize()
  File "/home/pi/led-stock-ticker/data/ticker.py", line 43, in initialize
    self.name = self.yf_ticker.info['shortName']
  File "/usr/local/lib/python3.9/dist-packages/yfinance/ticker.py", line 142, in                                                                                                              info
    return self.get_info()
  File "/usr/local/lib/python3.9/dist-packages/yfinance/base.py", line 1211, in                                                                                                              get_info
    data = self._quote.info
  File "/usr/local/lib/python3.9/dist-packages/yfinance/scrapers/quote.py", line                                                                                                              96, in info
    self._scrape(self.proxy)
  File "/usr/local/lib/python3.9/dist-packages/yfinance/scrapers/quote.py", line                                                                                                              125, in _scrape
    json_data = self._data.get_json_data_stores(proxy=proxy)
  File "/usr/local/lib/python3.9/dist-packages/yfinance/data.py", line 40, in wr                                                                                                             apped
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/yfinance/data.py", line 256, in g                                                                                                             et_json_data_stores
    stores = decrypt_cryptojs_aes_stores(data)
  File "/usr/local/lib/python3.9/dist-packages/yfinance/data.py", line 190, in d                                                                                                             ecrypt_cryptojs_aes_stores
    raise Exception("yfinance failed to decrypt Yahoo data response with hardcod                                                                                                             ed keys, contact developers")
Exception: yfinance failed to decrypt Yahoo data response with hardcoded keys, c                                                                                                             ontact developers
Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.9/dist-packages/multitasking/__init__.py", line 1                                                                                                             04, in _run_via_pool
    return callee(*args, **kwargs)
  File "/home/pi/led-stock-ticker/api/data.py", line 105, in fetch_crypto
    crypto = Crypto(symbol, currency)
  File "<string>", line 10, in __init__
  File "/home/pi/led-stock-ticker/data/ticker.py", line 31, in __post_init__
    self.initialize()
  File "/home/pi/led-stock-ticker/data/crypto.py", line 17, in initialize
    super(Crypto, self).initialize()
  File "/home/pi/led-stock-ticker/data/ticker.py", line 43, in initialize
    self.name = self.yf_ticker.info['shortName']
  File "/usr/local/lib/python3.9/dist-packages/yfinance/ticker.py", line 142, in                                                                                                              info
    return self.get_info()
  File "/usr/local/lib/python3.9/dist-packages/yfinance/base.py", line 1211, in                                                                                                              get_info
Exception in thread Thread-3:
    data = self._quote.info
  File "/usr/local/lib/python3.9/dist-packages/yfinance/scrapers/quote.py", line                                                                                                              96, in info
    self._scrape(self.proxy)
  File "/usr/local/lib/python3.9/dist-packages/yfinance/scrapers/quote.py", line                                                                                                              125, in _scrape
    json_data = self._data.get_json_data_stores(proxy=proxy)
  File "/usr/local/lib/python3.9/dist-packages/yfinance/data.py", line 40, in wr                                                                                                             apped
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/yfinance/data.py", line 256, in g                                                                                                             et_json_data_stores
[... 206 lines cut ...]
  File "/home/pi/led-stock-ticker/data/ticker.py", line 31, in __post_init__
    self.initialize()
  File "/home/pi/led-stock-ticker/data/forex.py", line 13, in initialize
    super(Forex, self).initialize()
  File "/home/pi/led-stock-ticker/data/ticker.py", line 43, in initialize
    self.name = self.yf_ticker.info['shortName']
  File "/usr/local/lib/python3.9/dist-packages/yfinance/ticker.py", line 142, in info
    return self.get_info()
  File "/usr/local/lib/python3.9/dist-packages/yfinance/base.py", line 1211, in get_info
    data = self._quote.info
  File "/usr/local/lib/python3.9/dist-packages/yfinance/scrapers/quote.py", line 96, in info
    self._scrape(self.proxy)
  File "/usr/local/lib/python3.9/dist-packages/yfinance/scrapers/quote.py", line 125, in _scrape
    json_data = self._data.get_json_data_stores(proxy=proxy)
  File "/usr/local/lib/python3.9/dist-packages/yfinance/data.py", line 40, in wrapped
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/yfinance/data.py", line 256, in get_json_data_stores
    stores = decrypt_cryptojs_aes_stores(data)
  File "/usr/local/lib/python3.9/dist-packages/yfinance/data.py", line 190, in decrypt_cryptojs_aes_stores
    raise Exception("yfinance failed to decrypt Yahoo data response with hardcoded keys, contact developers")
Exception: yfinance failed to decrypt Yahoo data response with hardcoded keys, contact developers

Thx to help me

see you

zhilly, 2023-01-28:
01/28/2023 01:16:23 PM DEBUG: Reading JSON file at matrix/coords/w64h32.json
01/28/2023 01:16:23 PM DEBUG: STREAM b'IHDR' 16 13
01/28/2023 01:16:23 PM DEBUG: STREAM b'IDAT' 41 385
01/28/2023 01:16:23 PM DEBUG: STREAM b'IHDR' 16 13
01/28/2023 01:16:23 PM DEBUG: STREAM b'IDAT' 41 1008
01/28/2023 01:16:23 PM DEBUG: Reading JSON file at matrix/config.schema.json
01/28/2023 01:16:23 PM DEBUG: Reading JSON file at matrix/config.json
01/28/2023 01:16:23 PM DEBUG: STREAM b'IHDR' 16 13
01/28/2023 01:16:23 PM DEBUG: STREAM b'sRGB' 41 1
01/28/2023 01:16:23 PM DEBUG: STREAM b'gAMA' 54 4
01/28/2023 01:16:23 PM DEBUG: STREAM b'pHYs' 70 9
01/28/2023 01:16:23 PM DEBUG: STREAM b'tEXt' 91 33
01/28/2023 01:16:23 PM DEBUG: STREAM b'IDAT' 136 13196
01/28/2023 01:16:23 PM INFO: Initializing data...
01/28/2023 01:16:24 PM DEBUG: Fetching initial data for TSLA.
01/28/2023 01:16:24 PM DEBUG: Fetching initial data for AMZN.
01/28/2023 01:16:24 PM DEBUG: Fetching initial data for MSFT.
01/28/2023 01:16:24 PM DEBUG: Fetching initial data for BTC-USD.
01/28/2023 01:16:24 PM DEBUG: Starting new HTTPS connection (1): finance.yahoo.com:443
01/28/2023 01:16:24 PM DEBUG: Starting new HTTPS connection (1): finance.yahoo.com:443
01/28/2023 01:16:24 PM DEBUG: Starting new HTTPS connection (1): finance.yahoo.com:443
01/28/2023 01:16:24 PM DEBUG: Starting new HTTPS connection (1): finance.yahoo.com:443
01/28/2023 01:16:25 PM DEBUG: https://finance.yahoo.com:443 "GET /quote/MSFT HTTP/1.1" 200 None
01/28/2023 01:16:25 PM DEBUG: https://finance.yahoo.com:443 "GET /quote/AMZN HTTP/1.1" 200 None
01/28/2023 01:16:25 PM DEBUG: https://finance.yahoo.com:443 "GET /quote/TSLA HTTP/1.1" 200 None
01/28/2023 01:16:25 PM DEBUG: https://finance.yahoo.com:443 "GET /quote/BTC-USD HTTP/1.1" 200 None
01/28/2023 01:16:29 PM DEBUG: Fetching initial data for ETH-USD.
01/28/2023 01:16:29 PM DEBUG: Starting new HTTPS connection (1): finance.yahoo.com:443
01/28/2023 01:16:29 PM DEBUG: Fetching initial data for LTC-USD.
01/28/2023 01:16:29 PM DEBUG: Starting new HTTPS connection (1): finance.yahoo.com:443
01/28/2023 01:16:29 PM DEBUG: Fetching initial data for USDEUR=X.
01/28/2023 01:16:29 PM DEBUG: Starting new HTTPS connection (1): finance.yahoo.com:443
01/28/2023 01:16:30 PM DEBUG: Fetching initial data for EURJPY=X.
01/28/2023 01:16:30 PM DEBUG: Starting new HTTPS connection (1): finance.yahoo.com:443
01/28/2023 01:16:30 PM DEBUG: https://finance.yahoo.com:443 "GET /quote/LTC-USD HTTP/1.1" 200 None
01/28/2023 01:16:30 PM DEBUG: https://finance.yahoo.com:443 "GET /quote/ETH-USD HTTP/1.1" 200 None
01/28/2023 01:16:30 PM DEBUG: https://finance.yahoo.com:443 "GET /quote/USDEUR=X HTTP/1.1" 200 None
01/28/2023 01:16:31 PM DEBUG: https://finance.yahoo.com:443 "GET /quote/EURJPY=X HTTP/1.1" 200 None
01/28/2023 01:16:33 PM DEBUG: Fetching initial data for GBPUSD=X.
01/28/2023 01:16:33 PM DEBUG: Starting new HTTPS connection (1): finance.yahoo.com:443
01/28/2023 01:16:38 PM DEBUG: https://finance.yahoo.com:443 "GET /quote/GBPUSD=X HTTP/1.1" 200 None

feram18, 2023-01-28:
Looks like more decryption issues came up since `yfinance` 0.2.5 version. I have pushed a [new release](https://github.com/feram18/led-stock-ticker/releases/tag/v0.3.7) updating the dependency. Issue should be fixed, apologies for the inconvenience.

closed 2023-01-28 by feram18

zhilly, 2023-01-29:
it works well now, thx you a lot you are the best!! have good we