feram18/led-stock-ticker #18: Stock ticker not working Yahoo finance data error

copyfile, 2023-02-17:
Since a couple of days the stock ticker is not working for me anymore. I get the below error messages related to Yahoo Finance. Any idea how to fix this?

    json_data = self._data.get_json_data_stores(proxy=proxy)
  File "/usr/local/lib/python3.9/dist-packages/yfinance/data.py", line 41, in wrapped
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/yfinance/data.py", line 307, in get_json_data_stores
    stores = decrypt_cryptojs_aes_stores(data, keys)
  File "/usr/local/lib/python3.9/dist-packages/yfinance/data.py", line 162, in decrypt_cryptojs_aes_stores
    raise Exception("yfinance failed to decrypt Yahoo data response")
Exception: yfinance failed to decrypt Yahoo data response
  File "/home/pi/led-stock-ticker/data/ticker.py", line 31, in __post_init__
    self.initialize()
  File "/home/pi/led-stock-ticker/data/stock.py", line 12, in initialize
    super(Stock, self).initialize()
  File "/home/pi/led-stock-ticker/data/ticker.py", line 43, in initialize
    self.name = self.yf_ticker.info['shortName']
  File "/usr/local/lib/python3.9/dist-packages/yfinance/ticker.py", line 142, in info
    return self.get_info()
  File "/usr/local/lib/python3.9/dist-packages/yfinance/base.py", line 1268, in get_info
    data = self._quote.info
  File "/usr/local/lib/python3.9/dist-packages/yfinance/scrapers/quote.py", line 95, in info
    self._scrape(self.proxy)
  File "/usr/local/lib/python3.9/dist-packages/yfinance/scrapers/quote.py", line 124, in _scrape
    json_data = self._data.get_json_data_stores(proxy=proxy)
  File "/usr/local/lib/python3.9/dist-packages/yfinance/data.py", line 41, in wrapped
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/yfinance/data.py", line 307, in get_json_data_stores
    stores = decrypt_cryptojs_aes_stores(data, keys)
  File "/usr/local/lib/python3.9/dist-packages/yfinance/data.py", line 162, in decrypt_cryptojs_aes_stores
    raise Exception("yfinance failed to decrypt Yahoo data response")
Exception: yfinance failed to decrypt Yahoo data response
Exception in thread Thread-13:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.9/dist-packages/multitasking/__init__.py", line 104, in _run_via_pool
    return callee(*args, **kwargs)
  File "/home/pi/led-stock-ticker/api/data.py", line 91, in fetch_stock
    stock = Stock(symbol, currency)
  File "<string>", line 10, in __init__
  File "/home/pi/led-stock-ticker/data/ticker.py", line 31, in __post_init__
    self.initialize()
  File "/home/pi/led-stock-ticker/data/stock.py", line 12, in initialize
    super(Stock, self).initialize()
  File "/home/pi/led-stock-ticker/data/ticker.py", line 43, in initialize
    self.name = self.yf_ticker.info['shortName']
  File "/usr/local/lib/python3.9/dist-packages/yfinance/ticker.py", line 142, in info
    return self.get_info()
  File "/usr/local/lib/python3.9/dist-packages/yfinance/base.py", line 1268, in get_info
    data = self._quote.info
  File "/usr/local/lib/python3.9/dist-packages/yfinance/scrapers/quote.py", line 95, in info
    self._scrape(self.proxy)
  File "/usr/local/lib/python3.9/dist-packages/yfinance/scrapers/quote.py", line 124, in _scrape
    json_data = self._data.get_json_data_stores(proxy=proxy)
  File "/usr/local/lib/python3.9/dist-packages/yfinance/data.py", line 41, in wrapped
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/yfinance/data.py", line 307, in get_json_data_stores
    stores = decrypt_cryptojs_aes_stores(data, keys)
  File "/usr/local/lib/python3.9/dist-packages/yfinance/data.py", line 162, in decrypt_cryptojs_aes_stores
    raise Exception("yfinance failed to decrypt Yahoo data response")
Exception: yfinance failed to decrypt Yahoo data response
Exception in thread Thread-14:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.9/dist-packages/multitasking/__init__.py", line 104, in _run_via_pool
[... 704 lines cut ...]
    return callee(*args, **kwargs)
  File "/home/pi/led-stock-ticker/api/data.py", line 118, in fetch_forex
    forex = Forex(symbol)
  File "<string>", line 9, in __init__
  File "/home/pi/led-stock-ticker/data/ticker.py", line 31, in __post_init__
    self.initialize()
  File "/home/pi/led-stock-ticker/data/forex.py", line 13, in initialize
    super(Forex, self).initialize()
  File "/home/pi/led-stock-ticker/data/ticker.py", line 43, in initialize
    self.name = self.yf_ticker.info['shortName']
  File "/usr/local/lib/python3.9/dist-packages/yfinance/ticker.py", line 142, in info
    return self.get_info()
  File "/usr/local/lib/python3.9/dist-packages/yfinance/base.py", line 1268, in get_info
    data = self._quote.info
  File "/usr/local/lib/python3.9/dist-packages/yfinance/scrapers/quote.py", line 95, in info
    self._scrape(self.proxy)
  File "/usr/local/lib/python3.9/dist-packages/yfinance/scrapers/quote.py", line 124, in _scrape
    json_data = self._data.get_json_data_stores(proxy=proxy)
  File "/usr/local/lib/python3.9/dist-packages/yfinance/data.py", line 41, in wrapped
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/yfinance/data.py", line 307, in get_json_data_stores
    stores = decrypt_cryptojs_aes_stores(data, keys)
  File "/usr/local/lib/python3.9/dist-packages/yfinance/data.py", line 162, in decrypt_cryptojs_aes_stores
    raise Exception("yfinance failed to decrypt Yahoo data response")
Exception: yfinance failed to decrypt Yahoo data response

feram18, 2023-02-17:
Unfortunately the `yfinance` dependency is having some issues due to changes in Yahoo's Finance site. They're still working to get it fixed ([see their discussion](https://github.com/ranaroussi/yfinance/issues/1407)), but at this moment there's no solution. I'm monitoring their discussion to apply any fix they come out with, but at the moment it appears complicated. I might have to implement another source for the data if they are unable to come up with a solution. Apologies for the inconvenience.

feram18, 2023-02-24:
I'm currently working on using an alternate data source to bring the project back up to some capacity.

feram18, 2023-02-25:
New release ([v0.3.8](https://github.com/feram18/led-stock-ticker/releases/tag/v0.3.8)) is up. Thank you for your patience.

closed 2023-02-25 by feram18

copyfile, 2023-02-25:
I just installed the latest version with the upgrade option but for me it is not working yet. See below errors. Is there anything else I need to do to make it work?

📊 LED-Stock-Ticker - v0.3.8 (64x32)
Exception in thread Thread-24:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.9/dist-packages/multitasking/__init__.py", line 104, in _run_via_pool
    return callee(*args, **kwargs)
  File "/home/pi/led-stock-ticker/api/data.py", line 91, in fetch_stock
    stock = Stock(symbol, currency)
  File "<string>", line 10, in __init__
  File "/home/pi/led-stock-ticker/data/ticker.py", line 31, in __post_init__
    self.initialize()
  File "/home/pi/led-stock-ticker/data/stock.py", line 12, in initialize
    super(Stock, self).initialize()
  File "/home/pi/led-stock-ticker/data/ticker.py", line 43, in initialize
    self.quote = self.yq_ticker.quotes.get(self.symbol.upper())
AttributeError: 'str' object has no attribute 'get'
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/requests/adapters.py", line 489, in send
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/requests/adapters.py", line 489, in send
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/requests/adapters.py", line 489, in send
    resp = conn.urlopen(
    resp = conn.urlopen(
    resp = conn.urlopen(
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 846, in urlopen
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 846, in urlopen
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 846, in urlopen
    return self.urlopen(
    return self.urlopen(
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 846, in urlopen
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 846, in urlopen
    return self.urlopen(
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 846, in urlopen
    return self.urlopen(
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 846, in urlopen
    return self.urlopen(
  [Previous line repeated 2 more times]
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 836, in urlopen
    return self.urlopen(
  [Previous line repeated 2 more times]
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 836, in urlopen
    retries = retries.increment(method, url, response=response, _pool=self)
  File "/usr/lib/python3/dist-packages/urllib3/util/retry.py", line 574, in increment
    retries = retries.increment(method, url, response=response, _pool=self)
  File "/usr/lib/python3/dist-packages/urllib3/util/retry.py", line 574, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='query2.finance.yahoo.com', port=443): Max retries exceeded with url: /v10/finance/quoteSummary/%5EIXIC?modules=summaryProfile&formatted=false&lang=en-US&region=US&corsDomain=finance.yahoo.com (Caused by ResponseError('too many 404 error responses'))
    raise MaxRetryError(_pool, url, error or ResponseError(cause))

During handling of the above exception, another exception occurred:

urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='query2.finance.yahoo.com', port=443): Max retries exceeded with url: /v10/finance/quoteSummary/%5EGSPC?modules=summaryProfile&formatted=false&lang=en-US&region=US&corsDomain=finance.yahoo.com (Caused by ResponseError('too many 404 error responses'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
[... 214 lines cut ...]
  File "/usr/local/lib/python3.9/dist-packages/yahooquery/base.py", line 1155, in _construct_urls
    return self.request("GET", url, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/requests/sessions.py", line 587, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python3.9/dist-packages/requests/sessions.py", line 701, in send
    urls = [
  File "/usr/local/lib/python3.9/dist-packages/yahooquery/base.py", line 1156, in <listcomp>
    r = adapter.send(request, **kwargs)
    self.session.get(
  File "/usr/local/lib/python3.9/dist-packages/yahooquery/utils/__init__.py", line 48, in send
  File "/usr/local/lib/python3.9/dist-packages/requests/sessions.py", line 600, in get
    return super(TimeoutHTTPAdapter, self).send(request, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/requests/adapters.py", line 556, in send
    raise RetryError(e, request=request)
requests.exceptions.RetryError: HTTPSConnectionPool(host='query2.finance.yahoo.com', port=443): Max retries exceeded with url: /v10/finance/quoteSummary/EURUSD=X?modules=summaryProfile&formatted=false&lang=en-US&region=US&corsDomain=finance.yahoo.com (Caused by ResponseError('too many 404 error responses'))
    return self.request("GET", url, **kwargs)
[... cut ...]
    return super(TimeoutHTTPAdapter, self).send(request, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/requests/adapters.py", line 556, in send
    raise RetryError(e, request=request)
requests.exceptions.RetryError: HTTPSConnectionPool(host='query2.finance.yahoo.com', port=443): Max retries exceeded with url: /v10/finance/quoteSummary/%5EAEX?modules=summaryProfile&formatted=false&lang=en-US&region=US&corsDomain=finance.yahoo.com (Caused by ResponseError('too many 404 error responses'))

feram18, 2023-02-25:
Can you share what your `config.json` looks like?

feram18, 2023-02-26:
Nevermind. The issue was caused when trying to get a URL for logo/image on non-equity tickers (Indices, ETFs, Mutual Funds, etc). It should be resolved now. Let me know if you encounter any other issues.

copyfile, 2023-02-26:
Awesome it's working now!! Thanks for all your help/support!!