feram18/led-stock-ticker #20: "€" symbol not working

Gautchh, 2023-03-29:
Hi,
You have made a great job on this project, all the code is well-structured.

I got some trouble with the "€" symbol, when I choose this symbol in the config.py. The matrix show only the logo and the clock, and the program stops. When I choose the "USD" everything work fine.
I try several things but nothing work... I got the error code : "UnicodeEncodeError: 'latin-1' codec can't encode character '\u20ac' in position 0: ordinal not in range(256)" in the led-stock-ticker.log.
If you have some time, can you help me find and correct this bug.

Have a nice day,
Gautchh

feram18, 2023-03-29:
I see the issue. It's not on your end. It looks like it's having an issue with rendering the character to the screen. The character might be missing from it. I'll have to fix the font file. Apologies for the inconvenience!

Gautchh, 2023-03-30:
Hi, Thank you to consider this issue. I don't need apologies, you do such a great work in this project.
Can you inform me if you have the time to correct this issue ?

Have a nice day,

feram18, 2023-04-04:
Issue should be fixed in [v0.3.10](https://github.com/feram18/led-stock-ticker/releases/tag/v0.3.10).

For anyone interested, the issue was caused because PIL fonts are limited to only 256 characters, so the € character (amongst many others) was not present in the PIL font, though it was present the original BDF font file. To resolve the issue, we are now using TTF fonts which are not limited.

closed 2023-04-04 by feram18