feram18/led-stock-ticker #21: Open/Close market indicator not working

Gautchh, 2023-04-12:
**Open/Close market indicator not working**
The indicator before the stock name is not working correctly. It may be red when the market is close and green when it is open.

**Configuration**
 - Raspberry Pi 3B+
 - Operative System: Raspberry Pi OS
 - LED Matrix Size: 128×64
 - Display Adapter: Adafruit RGB Matrix HAT
 - Led stock ticker v0.3.10

**Picture**
On this picture of the RUBIS stock, the indicator is red, and the picture was taken at 15:14 and the market close at 17:30.
[image]

**Additional context**
I'm in France, and I use the .PA (for the Paris stock) after the stock name to have the current price in France. It may cause the problem.

Your project is so great, I buy a new 128x64 RGB panel only for your project. Thank you !

feram18, 2023-04-13:
Thanks for bringing it up. When I started the project I was only thinking of NYSE so the market status indicator was following those hours instead of the individual stock's market.

I got a fix for it, pending further testing.

feram18, 2023-04-18:
Resolved in [v0.3.11](https://github.com/feram18/led-stock-ticker/releases/tag/v0.3.11).

closed 2023-04-18 by feram18

Gautchh, 2023-04-18:
Hi, Thanks. I will try this new version tomorrow.
Thank you for all your work.