feram18/led-stock-ticker #25: Function to run it at startup?

tobeehh, 2023-06-14:
Any chance to include an option to run this on startup in the config script as well? Alternatively do a how-to? I've had problems getting it to run at startup on a raspberry pi zero...

feram18, 2023-06-19:
I use a `systemd` service on my Pi for this.

1. Create the service file `nano led-stock-ticker.service`
2. Populate the service file with the contents below and replace (a) the parameters/flags that work for your setup in `TICKER_ARGS`, and (b) the working directory of the repo in `WorkingDirectory`
```
[Unit]
Description=LED Stock Ticker
Wants=network-online.target
After=network.target network-online.target

[Service]
Environment="TICKER_ARGS=--led-gpio-mapping=adafruit-hat --led-cols=64 --led-rows=32 --led-slowdown-gpio=4 --debug"
WorkingDirectory=/home/pi/led-stock-ticker
ExecStart=/usr/bin/python3 main.py $TICKER_ARGS
Restart=always
RestartSec=10

[Install]
WantedBy=multi-user.target
```
3. Save and move the service file `sudo mv led-stock-ticker.service /lib/systemd/system`
4. Enable the service `sudo systemctl enable led-stock-ticker.service`
5. Restart `sudo reboot`

This will have the project start up when the pi boots up, and also restart in the event of a crash.

To manually stop the service: `sudo service mlb-led-scoreboard stop`.
Use the same syntax to manually `start`, `restart`, or `status` check.

I'll create a wiki entry and update the README for others to see

tobeehh, 2023-06-20:
Thanks, worked like a charm. great Project.

closed 2023-06-20 by feram18