feram18/led-stock-ticker #26: Script crashes after some time

tobeehh, 2023-06-14:
Attached is the log:

> 06/14/2023 01:58:27 PM DEBUG: Incremented Retry for (url='/v6/finance/quote?lang=en-US&region=US&corsDomain=finance.yahoo.com&symbols=TSLA'): Retry(total=1, connect=None, re>
06/14/2023 01:58:30 PM DEBUG: Retry: /v6/finance/quote?lang=en-US&region=US&corsDomain=finance.yahoo.com&symbols=TSLA
06/14/2023 01:58:30 PM DEBUG: https://query2.finance.yahoo.com:443 "GET /v6/finance/quote?lang=en-US&region=US&corsDomain=finance.yahoo.com&symbols=TSLA HTTP/1.1" 404 91
06/14/2023 01:58:30 PM DEBUG: Incremented Retry for (url='/v6/finance/quote?lang=en-US&region=US&corsDomain=finance.yahoo.com&symbols=TSLA'): Retry(total=0, connect=None, re>
06/14/2023 01:58:34 PM DEBUG: Retry: /v6/finance/quote?lang=en-US&region=US&corsDomain=finance.yahoo.com&symbols=TSLA
06/14/2023 01:58:35 PM DEBUG: https://query2.finance.yahoo.com:443 "GET /v6/finance/quote?lang=en-US&region=US&corsDomain=finance.yahoo.com&symbols=TSLA HTTP/1.1" 404 91
06/14/2023 01:58:35 PM ERROR: HTTPSConnectionPool(host='query2.finance.yahoo.com', port=443): Max retries exceeded with url: /v6/finance/quote?lang=en-US&region=US&corsDomai>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 439, in send
    resp = conn.urlopen(
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 846, in urlopen
    return self.urlopen(
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 846, in urlopen
    return self.urlopen(
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 846, in urlopen
    return self.urlopen(
  [Previous line repeated 2 more times]
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 836, in urlopen
    retries = retries.increment(method, url, response=response, _pool=self)
  File "/usr/lib/python3/dist-packages/urllib3/util/retry.py", line 574, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='query2.finance.yahoo.com', port=443): Max retries exceeded with url: /v6/finance/quote?lang=en-US&region=US&corsD>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/Tobias/led-stock-ticker/main.py", line 48, in <module>
    main()
  File "/home/Tobias/led-stock-ticker/main.py", line 23, in main
    MainRenderer(matrix, canvas, draw, config, data)
  File "/home/Tobias/led-stock-ticker/renderer/main.py", line 35, in __init__
    self.render()
  File "/home/Tobias/led-stock-ticker/renderer/main.py", line 45, in render
    self.status = self.data.update()
  File "/home/Tobias/led-stock-ticker/api/data.py", line 66, in update
    self.update_ticker(ticker)
  File "/usr/local/lib/python3.9/dist-packages/multitasking/__init__.py", line 109, in async_method
    return callee(*args, **kwargs)
  File "/home/Tobias/led-stock-ticker/api/data.py", line 123, in update_ticker
    ticker.update()
  File "/home/Tobias/led-stock-ticker/data/stock.py", line 39, in update
    super(Stock, self).update()
  File "/home/Tobias/led-stock-ticker/data/ticker.py", line 70, in update
    self.price_data = self.yq_ticker.quotes.get(self.symbol.upper())
  File "/usr/local/lib/python3.9/dist-packages/yahooquery/ticker.py", line 418, in quotes
    data = self._chunk_symbols("quotes", list_result=True)
  File "/usr/local/lib/python3.9/dist-packages/yahooquery/base.py", line 999, in _chunk_symbols
    data = self._get_data(key, params, disable=True, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/yahooquery/base.py", line 1082, in _get_data
    urls = self._construct_urls(config, params, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/yahooquery/base.py", line 1148, in _construct_urls
    urls = [self.session.get(url=config["path"], params=params)]
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 555, in get
    return self.request('GET', url, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 542, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 655, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/yahooquery/utils/__init__.py", line 47, in send
    return super(TimeoutHTTPAdapter, self).send(request, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 507, in send
    raise RetryError(e, request=request)
requests.exceptions.RetryError: HTTPSConnectionPool(host='query2.finance.yahoo.com', port=443): Max retries exceeded with url: /v6/finance/quote?lang=en-US&region=US&corsDom>

feram18, 2023-06-19:
This looks like an issue from 0.3.11 that was fixed in the latest version (0.3.12). I'm not able to replicate the issue unless I go back a version. Are you sure you're in the latest version?

feram18, 2023-06-19:
Sorry, I just found the issue. I missed a change when fixing it in the last update. Should be fixed in [v0.3.13](https://github.com/feram18/led-stock-ticker/releases/tag/v0.3.13).

closed 2023-06-19 by feram18

tobeehh, 2023-06-20:
Works perfectly now, thank you so much!