feram18/led-stock-ticker #27: Financial data can't be collected, display stuck on clock

yoavzuri, 2023-07-14:
**Describe the bug**
The code doesn't work. Seems like yahooquery is broken.

**Configuration**
 - Raspberry Pi Model: 4B
 - Operating System: Raspbian 11
 - LED Matrix Size: 32x192 (different issue, I know :))
 - Display Adapter: Adafruit RGB Matrix HAT
 - Python Version: 3.9.2
 - Arguments/Flags used to execute script: --led-gpio-mapping=adafruit-hat --led-slowdown-gpio 2 --debug

Logs:
```
07/14/2023 07:32:23 PM DEBUG: Reading JSON file at matrix/coords/w64h32.json
07/14/2023 07:32:23 PM DEBUG: Reading JSON file at matrix/config.schema.json
07/14/2023 07:32:23 PM DEBUG: Reading JSON file at matrix/config.json
07/14/2023 07:32:23 PM DEBUG: STREAM b'IHDR' 16 13
07/14/2023 07:32:23 PM DEBUG: STREAM b'sRGB' 41 1
07/14/2023 07:32:23 PM DEBUG: STREAM b'gAMA' 54 4
07/14/2023 07:32:23 PM DEBUG: STREAM b'pHYs' 70 9
07/14/2023 07:32:23 PM DEBUG: STREAM b'tEXt' 91 33
07/14/2023 07:32:23 PM DEBUG: STREAM b'IDAT' 136 13196
07/14/2023 07:32:23 PM INFO: Initializing data...
07/14/2023 07:32:23 PM DEBUG: Fetching initial data for TSLA.
07/14/2023 07:32:23 PM DEBUG: Fetching initial data for SPY.
07/14/2023 07:32:23 PM DEBUG: Fetching initial data for MSFT.
07/14/2023 07:32:23 PM DEBUG: Starting new HTTPS connection (1): query2.finance.yahoo.com:443
07/14/2023 07:32:23 PM DEBUG: Starting new HTTPS connection (1): query2.finance.yahoo.com:443
07/14/2023 07:32:23 PM DEBUG: Starting new HTTPS connection (1): query2.finance.yahoo.com:443
07/14/2023 07:32:23 PM DEBUG: https://query2.finance.yahoo.com:443 "GET /v6/finance/quote/validate?lang=en-US&region=US&corsDomain=finance.yahoo.com&symbols=MSFT HTTP/1.1" 200 61
07/14/2023 07:32:23 PM DEBUG: https://query2.finance.yahoo.com:443 "GET /v6/finance/quote/validate?lang=en-US&region=US&corsDomain=finance.yahoo.com&symbols=SPY HTTP/1.1" 200 60
07/14/2023 07:32:23 PM DEBUG: https://query2.finance.yahoo.com:443 "GET /v6/finance/quote/validate?lang=en-US&region=US&corsDomain=finance.yahoo.com&symbols=TSLA HTTP/1.1" 200 61
07/14/2023 07:32:23 PM DEBUG: https://query2.finance.yahoo.com:443 "GET /v10/finance/quoteSummary/MSFT?modules=price&formatted=false&lang=en-US&region=US&corsDomain=finance.yahoo.com HTTP/1.1" 401 90
07/14/2023 07:32:23 PM ERROR: No data available for MSFT.
07/14/2023 07:32:23 PM WARNING: Stock: MSFT may not be valid.
07/14/2023 07:32:23 PM DEBUG: https://query2.finance.yahoo.com:443 "GET /v10/finance/quoteSummary/SPY?modules=price&formatted=false&lang=en-US&region=US&corsDomain=finance.yahoo.com HTTP/1.1" 401 90
07/14/2023 07:32:23 PM ERROR: No data available for SPY.
07/14/2023 07:32:23 PM WARNING: Stock: SPY may not be valid.
07/14/2023 07:32:23 PM DEBUG: https://query2.finance.yahoo.com:443 "GET /v10/finance/quoteSummary/TSLA?modules=price&formatted=false&lang=en-US&region=US&corsDomain=finance.yahoo.com HTTP/1.1" 401 90
07/14/2023 07:32:23 PM ERROR: No data available for TSLA.
07/14/2023 07:32:23 PM WARNING: Stock: TSLA may not be valid.
```

@feram18 regardless of the issue - super cool project!

yoavzuri, 2023-07-15:
Looks like yahooquery newest version fixed the issue.
yahooquery==2.3.1 -> yahooquery==2.3.2 in requirements.txt fixed it.

feram18, 2023-07-17:
Fixed. Thank you for reporting!

closed 2023-07-17 by feram18