feram18/led-stock-ticker #30: LED Matrix Stopped Updating

mattrcarroll, 2023-11-02:
Hi,

My Led Stock Ticker stopped updating prices on 10/31/23 and I have not been able to resolve.  It has been working fine for many months prior.  Any suggestions on how to resolve?   I have attached the log below.  This is an awesome project and I really appreciate you sharing it!  -Matt

[led-stock-ticker.log](https://github.com/feram18/led-stock-ticker/files/13232740/led-stock-ticker.log)

**Configuration**
 - Raspberry Pi Model: 3B+
 - Operative System: Raspberry Pi OS
 - LED Matrix Size: 64×32
 - Display Adapter: Adafruit RGB Matrix HAT 
 - Python Version: 3.9.2

feram18, 2023-11-02:
There seems to be an issue with the yahoo finance API. I'll investigate a solution.

mattrcarroll, 2023-11-08:
I updated the latest version of yahooquery (2.3.3) but that did not fix it.

feram18, 2023-11-12:
Only updating `yahooquery` to `2.3.3` did not work for me either, but I found that my network was blocking a domain required for that the API queries (`fc.yahoo.com`). [Others experienced this issue as well](https://github.com/dpguthrie/yahooquery/issues/233). 

After updating the dependency and whitelisting this domain I was able to get it working again.

mattrcarroll, 2023-11-13:
I was using quad9 for network dns which has some filtering so per your comment above on possible blocking of fc.yahoo.com I  switched to 1.1.1.1 which should be completely unfiltered.  However, this did not resolve the issue for me.

mattrcarroll, 2023-11-14:
Any other suggestions on how to troubleshoot or solve?  As discussed in Yahooquery (https://github.com/dpguthrie/yahooquery/issues/224), version 2.3.3 solves it for some, but not others.  Or, 2.3.3 solves it for a while but then eventually stops working.

mattrcarroll, 2023-11-19:
Configuration:

Raspberry Pi Model: 3B+ 
Operative System: Raspberry Pi OS (2023-05-03-raspios-bullseye-arm64-lite.img)
LED Matrix Size: 64×32
Display Adapter: Adafruit RGB Matrix HAT
Python Version: 3.9.2
YahooQuery: 2.3.3

I continue to have difficulty getting my LED Stock Ticker to work since the changes to the yahoo api were made.  Prior to this it worked perfectly. 

As suggested, I have updated to yahooquery 2.3.3 and confirmed there is no adblocking or dns filtering blocking fc.yahoo.com.  Still, receiving 404 errors.  

Interestingly though, yahooquery works just fine running in Python on a PC on the same network.  So, the isssue must be with some configuration on the raspberry pi.

To help rule out any issues with my existing configuration, I installed a new clean image of Raspberry Pi OS (2023-05-03-raspios-bullseye-arm64-lite.img) on the microsd card **and ran the below exact sequence of commands to install LED Stock Ticker, upgrade to Yahooquery 2.3.3, and then run main.py.**   

The log file showing the errors generated when main.py is run after running the below sequence of commands is attached.   
[led-stock-ticker.log](https://github.com/feram18/led-stock-ticker/files/13401996/led-stock-ticker.log)

Any suggestions or thoughts on what I am doing wrong or what the issue could be? 

----
Started with clean install of Raspberry Pi OS (2023-05-03-raspios-bullseye-arm64-lite.img).  Then did the following:

sudo apt-get update
sudo apt-get upgrade
sudo apt-get install git python3-pip -y

git clone --recursive https://github.com/feram18/led-stock-ticker.git
cd led-stock-ticker
chmod +x install.sh
./install.sh

cd led-stock-ticker
./update.sh

cd
pip3 install yahooquery –upgrade

cd led-stock-ticker
sudo python3 main.py --led-gpio-mapping="adafruit-hat" --led-slowdown-gpio=2 --debug

mattrcarroll, 2023-11-22:
> Only updating `yahooquery` to `2.3.3` did not work for me either, but I found that my network was blocking a domain that the API queries (`fc.yahoo.com`). [Others experienced this issue as well](https://github.com/dpguthrie/yahooquery/issues/233).
> 
> After updating the dependency and whitelisting this domain I was able to get it working again.

Hi Luis.  Did you simply do pip3 install yahooquery==2.3.3 to get yours working again?  I still show 404 errors in the log after making this update.  I use 1.1.1.1 for dns and don't use a network ad blocker.  Thank you for your help!

feram18, 2023-12-08:
Sorry for the delayed resolution.

Originally, only upgrading the `yahooquery` dependency worked on my dev setup, but later found out it did not fix the issue when running on my Raspberry Pi.
Upgraded a few other dependencies and now everything seems to be working on both my dev setup & two Raspberry Pi's.
Using strict dependency versions (now present in [v0.3.16](https://github.com/feram18/led-stock-ticker/releases/tag/v0.3.16)) should help with inter-dependency issues. Let me know if this works for you.

LiminalD, 2023-12-09:
Hi Feram, I installed the update but after the start-up screen it permanently switches to the clock...

Great project btw and keep up the good work!

Ulfb3rth, 2023-12-09:
Hello everyone, 
I have the same problem, only the clock is displayed 

[led-stock-ticker.log](https://github.com/feram18/led-stock-ticker/files/13624949/led-stock-ticker.log)

mattrcarroll, 2023-12-09:
The following solved the stuck on clock issue for me.  I am not an expert so may want to wait for @feram18 to weigh in on before running.

sudo apt install libopenjp2-7 libopenjp2-7-dev libopenjp2-tools
sudo apt-get install libopenblas-dev

LiminalD, 2023-12-11:
It seems that Yahoo is geo blocking and users outside the US are not able to retrieve data anymore... see also: https://github.com/dpguthrie/yahooquery/issues/241 Maybe a new feature request to choose between multiple data sources?

feram18, 2023-12-13:
> The following solved the stuck on clock issue for me. I am not an expert so may want to wait for @feram18 to weigh in on before running.
> 
> sudo apt install libopenjp2-7 libopenjp2-7-dev libopenjp2-tools sudo apt-get install libopenblas-dev

Perhaps the upgraded dependencies require these transient dependencies that your system did not yet have.

feram18, 2023-12-29:
> It seems that Yahoo is geo blocking and users outside the US are not able to retrieve data anymore... see also: [dpguthrie/yahooquery#241](https://github.com/dpguthrie/yahooquery/issues/241) Maybe a new feature request to choose between multiple data sources?

[Release 0.3.17](https://github.com/feram18/led-stock-ticker/releases/tag/v0.3.17) upgrades `yahooquery` to resolve this issue.

LiminalD, 2023-12-29:
Working again, great job!

closed 2023-12-31 by feram18