feram18/led-stock-ticker #31: Error import Numpy

rvloenhout, 2023-11-10:
Heyhey, Thank you very much for sharing this code!
I am running into a problem, my experience with Python is not super advance so if it is a sally mistake I am sorry in advance!
When running the main.py from the led-stock-ticker directory with sudo privilledges I get the following error:

Traceback (most recent call last):
  File "/home/root/led-stock-ticker/main.py", line 9, in <module>
    from api.data import Data
  File "/home/root/led-stock-ticker/api/data.py", line 9, in <module>
    from data.crypto import Crypto
  File "/home/root/led-stock-ticker/data/crypto.py", line 5, in <module>
    from data.ticker import Ticker
  File "/home/root/led-stock-ticker/data/ticker.py", line 5, in <module>
    import yahooquery
  File "/usr/local/lib/python3.9/dist-packages/yahooquery/__init__.py", line 6, in <module>
    from .research import Research  # noqa
  File "/usr/local/lib/python3.9/dist-packages/yahooquery/research.py", line 4, in <module>
    import pandas as pd
  File "/usr/local/lib/python3.9/dist-packages/pandas/__init__.py", line 16, in <module>
    raise ImportError(
ImportError: Unable to import required dependencies:
numpy: Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there.

------------------------
I have an Raspberry Pi 2B with Raspberry Pi OS.
I have updated everything by running: sudo apt-get update
Installed the dependancies and ran ./update.sh
I get this error with and without supplying flags.

If more information is needed I'd be happy to supply it.
Thank you in advance for you help!

rvloenhout, 2023-11-10:
I seem to have fixed the problem, a new version of Numpy was release in October (1.26.1). I downgraded to 1.25.2 and this seemed to do the trick. Do you want me to close the issue?

btkircher, 2023-11-10:
You're a life saver @rvloenhout . Just ran into this issue while setting up my Matrix.

mattrcarroll, 2023-11-10:
Is your matrix now working?  Mine stopped working on 10/31 and I thought it was related to an issue with yahooquery.  Maybe Numpy is the problem?

rvloenhout, 2023-11-11:
@mattrcarroll What error are you getting? Because I was getting an Yahoo Qeury time out error while testing, something around a 404 too many queries

Ulfb3rth, 2023-11-12:
Hello everyone,

I have the same error. But I also noticed an error during the installation. Does it look like this is related?

Do any of you have an idea what is wrong with the installation?

The display of the LED MAtrix via the /rpi-rgb-led-matrix/examples-api-use demo works 

Greetings Ulfberth 

 Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
    Preparing wheel metadata ... error
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 /tmp/tmp0u28mam8_in_process.py prepare_metadata_for_build_wheel /tmp/tmppc3umtyd
         cwd: /tmp/pip-install-ee2bv99t/pandas_0ace20314c074cf8aed0260ba9bf7064
    Complete output (31 lines):
    + meson setup /tmp/pip-install-ee2bv99t/pandas_0ace20314c074cf8aed0260ba9bf7064 /tmp/pip-install-ee2bv99t/pandas_0ace20314c074cf8aed0260ba9bf7064/.mesonpy-qrzxyute/build -Dbuildtype=release -Db_ndebug=if-release -Db_vscrt=md --vsenv --native-file=/tmp/pip-install-ee2bv99t/pandas_0ace20314c074cf8aed0260ba9bf7064/.mesonpy-qrzxyute/build/meson-python-native-file.ini
    The Meson build system
    Version: 1.2.1
    Source dir: /tmp/pip-install-ee2bv99t/pandas_0ace20314c074cf8aed0260ba9bf7064
    Build dir: /tmp/pip-install-ee2bv99t/pandas_0ace20314c074cf8aed0260ba9bf7064/.mesonpy-qrzxyute/build
    Build type: native build
    Project name: pandas
    Project version: 2.1.3
    C compiler for the host machine: cc (gcc 10.2.1 "cc (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110")
    C linker for the host machine: cc ld.bfd 2.35.2
    C++ compiler for the host machine: c++ (gcc 10.2.1 "c++ (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110")
    C++ linker for the host machine: c++ ld.bfd 2.35.2
    Cython compiler for the host machine: cython (cython 0.29.36)
    Host machine cpu family: arm
    Host machine cpu: armv7l
    Program python found: YES (/usr/bin/python3)
    
    ../../pandas/meson.build:1:15: ERROR: Command `/usr/bin/python3 -c '
    import os
    import numpy as np
    try:
        # Check if include directory is inside the pandas dir
        # e.g. a venv created inside the pandas dir
        # If so, convert it to a relative path
        incdir = os.path.relpath(np.get_include())
    except Exception:
        incdir = np.get_include()
    print(incdir)
         '` failed with status 1.
    
    A full log can be found at /tmp/pip-install-ee2bv99t/pandas_0ace20314c074cf8aed0260ba9bf7064/.mesonpy-qrzxyute/build/meson-logs/meson-log.txt
    ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/86/ff/662dde2193fc93b8547b073db20472b9676f944d907247a46c9c5bc45bfc/pandas-2.1.3.tar.gz#sha256=22929f84bca106921917eb73c1521317ddd0a4c71b395bcf767a106e3494209f (from https://pypi.org/simple/pandas/) (requires-python:>=3.9). Command errored out with exit status 1: /usr/bin/python3 /tmp/tmp0u28mam8_in_process.py prepare_metadata_for_build_wheel /tmp/tmppc3umtyd Check the logs for full command output.
  Downloading https://www.piwheels.org/simple/pandas/pandas-2.1.2-cp39-cp39-linux_armv7l.whl (41.8 MB)
     |████████████████████████████████| 41.8 MB 12 kB/s 
Collecting pillow==9.3.0
  Downloading https://www.piwheels.org/simple/pillow/Pillow-9.3.0-cp39-cp39-linux_armv7l.whl (1.1 MB)
     |████████████████████████████████| 1.1 MB 640 kB/s 
Collecting pytz>=2021.1
  Downloading https://www.piwheels.org/simple/pytz/pytz-2023.3.post1-py3-none-any.whl (502 kB)
     |████████████████████████████████| 502 kB 981 kB/s 
Collecting questionary>=1.10.0
  Downloading https://www.piwheels.org/simple/questionary/questionary-2.0.1-py3-none-any.whl (34 kB)
Requirement already satisfied: requests>=2.22.0 in /usr/lib/python3/dist-packages (from -r requirements.txt (line 9)) (2.25.1)
Collecting wheel>=0.37.0
  Using cached https://www.piwheels.org/simple/wheel/wheel-0.41.3-py3-none-any.whl (65 kB)
Collecting yahooquery==2.3.2
  Downloading https://www.piwheels.org/simple/yahooquery/yahooquery-2.3.2-py2.py3-none-any.whl (49 kB)
     |████████████████████████████████| 49 kB 529 kB/s 
Collecting requests-futures>=1.0.0
  Downloading https://www.piwheels.org/simple/requests-futures/requests_futures-1.0.1-py2.py3-none-any.whl (7.6 kB)
Collecting lxml>=4.5.1
[... 24 lines cut ...]
    Found existing installation: python-dateutil 2.8.1
    Not uninstalling python-dateutil at /usr/lib/python3/dist-packages, outside environment /usr
    Can't uninstall 'python-dateutil'. No files were found to uninstall.
  Attempting uninstall: numpy
    Found existing installation: numpy 1.19.5
    Not uninstalling numpy at /usr/lib/python3/dist-packages, outside environment /usr
    Can't uninstall 'numpy'. No files were found to uninstall.
  Attempting uninstall: lxml
    Found existing installation: lxml 4.6.3
    Not uninstalling lxml at /usr/lib/python3/dist-packages, outside environment /usr
    Can't uninstall 'lxml'. No files were found to uninstall.
  Attempting uninstall: wheel
    Found existing installation: wheel 0.34.2
    Not uninstalling wheel at /usr/lib/python3/dist-packages, outside environment /usr
    Can't uninstall 'wheel'. No files were found to uninstall.
  Attempting uninstall: pillow
    Found existing installation: Pillow 8.1.2
    Not uninstalling pillow at /usr/lib/python3/dist-packages, outside environment /usr
    Can't uninstall 'Pillow'. No files were found to uninstall.
Successfully installed attrs-23.1.0 jsonschema-4.16.0 lxml-4.9.3 multitasking-0.0.11 numpy-1.26.1 pandas-2.1.2 pillow-9.3.0 pyrsistent-0.20.0 python-dateutil-2.8.2 pytz-2023.3.post1 questionary-2.0.1 requests-futures-1.0.1 tqdm-4.66.1 tzdata-2023.3 wheel-0.41.3 yahooquery-2.3.2
Creating new config.json file...
Set your preferences using the command ./config.py
Update completed

If there are no errors shown above, installation was successful.

mattrcarroll, 2023-11-28:
> @mattrcarroll What error are you getting? Because I was getting an Yahoo Qeury time out error while testing, something around a 404 too many queries

[rvloenhout](https://github.com/rvloenhout) I am receving 404 errors.  The log is attached.   Are you having the same issue? 
 Thank you for your help.  

[led-stock-ticker.log](https://github.com/feram18/led-stock-ticker/files/13483665/led-stock-ticker.log)

rvloenhout, 2023-11-28:
I think this is the same problem, it seemed to be a problem with the API.
On stackoverflow people also mentioned problems with the yfinance API and after updating this might have fixed it.
Now I see this program uses yahooquery as their API and they released a new version today (2.3.3, requirements uses version 2.3.2), I wanted to test updating this tonight and see if this fixes the problem as I think it has to do this the Yahoo API and some changes made on their end.

rvloenhout, 2023-11-28:
Upgrading the package unfortunatly did not help, I will continue to search and try solutions

The errors I am receiving are:

led-stock-ticker.log

11/28/2023 04:44:53 PM DEBUG: Reading JSON file at matrix/coords/w64h32.json
11/28/2023 04:44:53 PM DEBUG: Reading JSON file at matrix/config.schema.json
11/28/2023 04:44:53 PM DEBUG: Reading JSON file at matrix/config.json
11/28/2023 04:44:53 PM DEBUG: STREAM b'IHDR' 16 13
11/28/2023 04:44:53 PM DEBUG: STREAM b'sRGB' 41 1
11/28/2023 04:44:53 PM DEBUG: STREAM b'gAMA' 54 4
11/28/2023 04:44:53 PM DEBUG: STREAM b'pHYs' 70 9
11/28/2023 04:44:53 PM DEBUG: STREAM b'tEXt' 91 33
11/28/2023 04:44:53 PM DEBUG: STREAM b'IDAT' 136 13196
11/28/2023 04:44:53 PM INFO: Initializing data...
11/28/2023 04:44:53 PM DEBUG: Fetching initial data for AAPL.
11/28/2023 04:44:53 PM DEBUG: Starting new HTTPS connection (1): query2.finance.yahoo.com:443
11/28/2023 04:44:53 PM DEBUG: https://query2.finance.yahoo.com:443 "GET /v6/finance/quote/validate?lang=en-US&region=US&co>
11/28/2023 04:44:53 PM DEBUG: https://query2.finance.yahoo.com:443 "GET /v6/finance/quoteSummary/AAPL?modules=price&format>
11/28/2023 04:44:53 PM DEBUG: Incremented Retry for (url='/v6/finance/quoteSummary/AAPL?modules=price&formatted=false&lang>
11/28/2023 04:44:53 PM DEBUG: Retry: /v6/finance/quoteSummary/AAPL?modules=price&formatted=false&lang=en-US&region=US&cors>
11/28/2023 04:44:53 PM DEBUG: https://query2.finance.yahoo.com:443 "GET /v6/finance/quoteSummary/AAPL?modules=price&format>
11/28/2023 04:44:53 PM DEBUG: Incremented Retry for (url='/v6/finance/quoteSummary/AAPL?modules=price&formatted=false&lang>
11/28/2023 04:44:54 PM DEBUG: Retry: /v6/finance/quoteSummary/AAPL?modules=price&formatted=false&lang=en-US&region=US&cors>
11/28/2023 04:44:54 PM DEBUG: https://query2.finance.yahoo.com:443 "GET /v6/finance/quoteSummary/AAPL?modules=price&format>
11/28/2023 04:44:54 PM DEBUG: Incremented Retry for (url='/v6/finance/quoteSummary/AAPL?modules=price&formatted=false&lang>
11/28/2023 04:44:55 PM DEBUG: Retry: /v6/finance/quoteSummary/AAPL?modules=price&formatted=false&lang=en-US&region=US&cors>
11/28/2023 04:44:55 PM DEBUG: https://query2.finance.yahoo.com:443 "GET /v6/finance/quoteSummary/AAPL?modules=price&format>
11/28/2023 04:44:55 PM DEBUG: Incremented Retry for (url='/v6/finance/quoteSummary/AAPL?modules=price&formatted=false&lang>
11/28/2023 04:44:58 PM DEBUG: Retry: /v6/finance/quoteSummary/AAPL?modules=price&formatted=false&lang=en-US&region=US&cors>
11/28/2023 04:44:58 PM DEBUG: https://query2.finance.yahoo.com:443 "GET /v6/finance/quoteSummary/AAPL?modules=price&format>
11/28/2023 04:44:58 PM DEBUG: Incremented Retry for (url='/v6/finance/quoteSummary/AAPL?modules=price&formatted=false&lang>
11/28/2023 04:45:03 PM DEBUG: Retry: /v6/finance/quoteSummary/AAPL?modules=price&formatted=false&lang=en-US&region=US&cors>
11/28/2023 04:45:03 PM DEBUG: https://query2.finance.yahoo.com:443 "GET /v6/finance/quoteSummary/AAPL?modules=price&format>

Error in command line:

Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 439, in send
    resp = conn.urlopen(
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 846, in                                            urlopen
    return self.urlopen(
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 846, in                                            urlopen
    return self.urlopen(
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 846, in                                            urlopen
    return self.urlopen(
  [Previous line repeated 2 more times]
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 836, in                                            urlopen
    retries = retries.increment(method, url, response=response, _pool=self)
  File "/usr/lib/python3/dist-packages/urllib3/util/retry.py", line 574, in incr                                           ement
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='query2.finance.yahoo                                           .com', port=443): Max retries exceeded with url: /v6/finance/quoteSummary/AAPL?m                                           odules=price&formatted=false&lang=en-US&region=US&corsDomain=finance.yahoo.com (                                           Caused by ResponseError('too many 404 error responses'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.9/dist-packages/multitasking/__init__.py", line 1                                           04, in _run_via_pool
    return callee(*args, **kwargs)
  File "/home/rob/led-stock-ticker/api/data.py", line 83, in fetch_stock
    stock = Stock(symbol, currency)
  File "<string>", line 11, in __init__
  File "/home/rob/led-stock-ticker/data/ticker.py", line 32, in __post_init__
    self.initialize()
  File "/home/rob/led-stock-ticker/data/stock.py", line 17, in initialize
    super(Stock, self).initialize()
  File "/home/rob/led-stock-ticker/data/ticker.py", line 52, in initialize
[... cut ...]
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 507, in send
    raise RetryError(e, request=request)
requests.exceptions.RetryError: HTTPSConnectionPool(host='query2.finance.yahoo.c                                           om', port=443): Max retries exceeded with url: /v6/finance/quoteSummary/AAPL?mod                                           ules=price&formatted=false&lang=en-US&region=US&corsDomain=finance.yahoo.com (Ca                                           used by ResponseError('too many 404 error responses'))

mattrcarroll, 2023-12-08:
I am also receiving the error: 

numpy: Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there.
 
I downgrade Numpy to 1.25.2 as suggested above but still continue to receive the error.

Any suggestions on how to resolve? 

Thank you!

feram18, 2023-12-08:
The latest `yahooquery` release should address the 404 errors. I posted a new release [v0.3.16](https://github.com/feram18/led-stock-ticker/releases/tag/v0.3.16) which should also resolve other inter-dependency issues. Let me know if this resolves the problems you are all experiencing.

feram18, 2023-12-08:
Also worth noting, updated dependencies required a minimum Python version 3.9.

rvloenhout, 2023-12-09:
This did fix the 404 errors! I just realized my display isnt compatible so I can verify if it completely works but I am not getting anymore errors :)

closed 2023-12-31 by feram18