feram18/led-stock-ticker #33: The ticker starts but only goes up to the the time and date and does not display anything there after. Please could you tell me how to resolve please?

DJsupreme98, 2023-12-14:
**Describe the bug**
A clear and concise description of what the bug is.

**Configuration**
 - Raspberry Pi Model: [e.g. 4B, 4A, 3B+, Zero W, etc.]
 - Operative System: [e.g. Raspberry Pi OS]
 - LED Matrix Size: [Supported Sizes: (32×16), (64×32), (128×64)]
 - Display Adapter: [e.g. Adafruit RGB Matrix HAT, Bonnet, etc.]
 - Python Version: [Python 3.7 or newer is required]
 - Arguments/Flags used to execute script: 

**Screenshots/Logs**
If applicable, add screenshots or logs to help explain your problem.

**Additional context**
Add any other information about the problem, such as your `config.json` file or any change you made prior to the issue.

feram18, 2023-12-29:
I believe any API querying issues should now be resolved by the [latest release 0.3.17](https://github.com/feram18/led-stock-ticker/releases/tag/v0.3.17).

DJsupreme98, 2023-12-29:
Hi Luis,

Thanks it is working again.

closed 2023-12-31 by feram18