feram18/led-stock-ticker #37: How do I attach the 128x64 dual board with two ribbons to the pi?

Jord4n, 2024-05-05:
I see hats with only 1 ribbon connector port and another with 3 connectors but no external power.  A bit confused. at this LED board I have has two connector ribbons

feram18, 2024-05-09:
By dual board, do you mean you have two 64x64 boards? or two 128x64?

Each board should have two ribbon connectors at the back, you can daisy chain the boards using the ribbon cable. 
One of the boards will be connected to the Pi hat, this board will have a power connector and will share power to the second board. The second board which is only connected to the first board does not need to be connected to the hat. Kind of [like this](https://there.oughta.be/assets/resized/images/2020-09-28/1280/daisychain.jpg).

Jord4n, 2024-05-19:
Ah!  I see!  Thank you so much, I missed that bit!  Thank you!

closed 2024-05-19 by Jord4n