feram18/led-stock-ticker #41: Experiencing fatal error in install script.

stray77, 2025-01-15:
**Describe the bug**
Experiencing fatal error in install script.

**Configuration**
 - Raspberry Pi Model:  Raspberry Pi Zero 2 W Rev1.0
 - Operative System:  Debian GNU/Linux 12 (bookworm) aarch64
 - LED Matrix Size: Waveshare 64x64 x2 = 128x64
 - Display Adapter:  Bonnet
 - Python Version: Python 3.11.2
 - Arguments/Flags used to execute script: Just copied and pasted the instructions:
 
git clone --recursive https://github.com/feram18/led-stock-ticker.git
cd led-stock-ticker
chmod +x install.sh
./install.sh

**Screenshots/Logs**
[LED Stock Ticker.txt](https://github.com/user-attachments/files/18426352/LED.Stock.Ticker.txt)
 
[image]

**Additional context**
No changes made.

PatrickFist, 2025-03-07:
I can confirm same error on raspberry 4 with raspbian lite, up to date as of today.

pi@raspberrypi:~/led-stock-ticker $  python3 --version
Python 3.11.2

amecreate, 2025-03-17:
Seems similar to this issue: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1526

My raspberry pi 3B is running python 3.7 on Linux buster. I bypassed install.sh, manually installed rpi-rgb-led-matrix,  ran update.sh with all dependencies downgraded to match the python version, then finally able to run `raspi-config` and `main.py`.

feram18, 2025-03-18:
I believe the current commit of the rpi-rgb-led-matrix library on this project just doesn't support Python 3.11. I think simply updating the library should resolve it. I'll take a look.