feram18/led-stock-ticker #47: The Ticker hangs in the loading screen

Ulfb3rth, 2025-06-08:
Thanks Ferma18 great project, the ticker is on daily.

Since 1 month the ticker hangs in the loading screen with the version of the ticker.
I just updated today to the latest version v0.1.18. The error persists. 
Then I have to switch the Raspberry pi off and on several times. Most nights around 4 the connection to yahooquery is established.
does anyone know the problem 

greetings Ulfb3rth

Error Code Log 

06/08/2025 06:04:31 PM WARNING: 1. try:
06/08/2025 06:05:00 PM CRITICAL: Failed to obtain crumb.  Ability to retrieve data will be significantly limited.
06/08/2025 06:05:09 PM CRITICAL: Failed to obtain crumb.  Ability to retrieve data will be significantly limited.
06/08/2025 06:07:08 PM WARNING: 1. try:
06/08/2025 06:07:37 PM CRITICAL: Failed to obtain crumb.  Ability to retrieve data will be significantly limited.
06/08/2025 06:07:38 PM CRITICAL: Failed to obtain crumb.  Ability to retrieve data will be significantly limited.
06/08/2025 06:07:47 PM CRITICAL: Failed to obtain crumb.  Ability to retrieve data will be significantly limited.
06/08/2025 06:07:54 PM CRITICAL: Failed to obtain crumb.  Ability to retrieve data will be significantly limited.
06/08/2025 06:08:25 PM WARNING: 1. try:
06/08/2025 06:10:15 PM CRITICAL: Failed to obtain crumb.  Ability to retrieve data will be significantly limited.
06/08/2025 06:20:28 PM WARNING: 1. try:
06/08/2025 06:20:34 PM WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='finance.yahoo.com', port=443): Read timed out. (read timeout=5)")': /
06/08/2025 06:22:56 PM CRITICAL: Failed to obtain crumb.  Ability to retrieve data will be significantly limited.
06/08/2025 07:17:07 PM WARNING: 1. try:
06/08/2025 07:21:35 PM WARNING: 1. try:
06/08/2025 07:35:08 PM WARNING: 1. try:
06/08/2025 07:37:28 PM WARNING: 1. try:

Benminnn, 2025-06-25:
I'm not getting a rate limit hit on 0.3.19:

requests.exceptions.RetryError: HTTPSConnectionPool(host='query2.finance.yahoo.com', port=443): Max retries exceeded with url: /v6/finance/quote/validate?lang=en-US&region=US&corsDomain=finance.yahoo.com&crumb=5yDqYIikhVU&symbols=INTC (Caused by ResponseError('too many 429 error responses'))

Benminnn, 2025-06-29:
simply needed to get everything updated with update.sh and runs great now, thank you